raspibolt/raspibolt #515: connecting electrum to my node running electrs

danyR85, 2019-12-19:
Hello,

I would appreciate some help regarding this issue.
electrum can't seem to connect to my node running electrs.

bitcoind and electrs are running fine.

I previously ran eps on my node (using the rpi3 raspibolt guide on rpi4) but I would like to change to electrs
Do I need to reindex the blockchain?

danyR85, 2019-12-21:
Reindexing didn't seem to work....
Electrs seem to run fine on the pi but electrum on my pc (windows) does not see it.

I'll try getting eps started.

PS:

It seems that tor didn't work either (ip was visible on bitcoin-cli getnetworkinfo | grep local)
until I added  "bind=127.0.0.1" to bitcoin.conf

danyR85, 2019-12-21:
Seems to be a Windows Issue?
I installed electrum on linux vm and it connected to electrs without issue

Stadicus, 2019-12-29:
The only thing different on Windows / Linux I can imaginge is the mDNS part, using `raspibolt.local` instead of the IP address. Otherwise Electrum should be platform independent. If it works on Linux, it should work on Windows, at least from the RaspiBolt side.

Stadicus, 2019-12-29:
> It seems that tor didn't work either (ip was visible on bitcoin-cli getnetworkinfo | grep local)
until I added "bind=127.0.0.1" to bitcoin.conf

This is not necessary on a new setup, but I guess other bitcoin nodes remember your old ip address and connect from their end over clearnet. To make the setup more robust, I added the `bind` configuration in https://github.com/Stadicus/RaspiBolt/pull/522

danyR85, 2020-01-11:
Figured it out.

In the Electrum datadir (%appdata%/electrum) there is a certs folder.
There was an old file there named as my node ip - I assume from when I was connecting to EPS.

I moved it to a different folder, and electrum connected to the node without issue.

closed 2020-01-11 by danyR85