raspibolt/raspibolt #518: Newbie Question/Issue: Making the node local only?

BluePixel319, 2019-12-26:
Hi All, how would I go about making sure my node can't be accessed from the outside? 

Would not installing TOR be enough? Would I need to modify any UFW strings? i.e.
`ufw allow from 192.168.0.0/24 to any port 22 `

I don't know enough to know what exactly is being opened up with the ufw/TOR steps, and am worried that machines from outside my LAN would be able to access the raspibolt.

spookiestevie, 2019-12-28:
I think you are all good. 
we previously set `ufw default deny incoming` this denies everything coming in
then `ufw allow from 192.168.0.0/24 to any port 22 ` is only allowing you to ssh within your network as the ip `192.168.0.0` basically means the local network only.

BluePixel319, 2019-12-30:
Thanks for the response. The guide says `$ ufw allow 22    comment 'allow SSH from LAN'`, but I didn't see how it specified LAN only, and the status says that the traffic would be allowed from anywhere? So I made that modification, I'll stick with it.

Final question, are there any risks related to the opening of the other three ports mentioned in the guide? Not sure if making these local only would break something (assuming I'm only ever planning on using this on LAN.

```
$ ufw allow 8333  comment 'allow Bitcoin'
$ ufw allow 9735  comment 'allow Lightning'
$ ufw allow 50002 comment 'allow Electrum SSL'
```

spookiestevie, 2019-12-30:
In short, it will be fine.

**But,**

> An open port is an attack surface
https://security.stackexchange.com/a/78811

I'm not a security expert btw but i will say:

If you don't have the skill to study the software you are running to make your own informed decision, you have to _trust_ that the service is not vulnerable to a buffer overflow, or another remotely exploitable vulnerability.

You have to assess through external observation, are the bitcoin and lnd devs going anywhere, how much peer review is happening, you may even need to self reflect, are you deemed a high value target (ie exchange), you can also assume [lindy effect](https://en.wikipedia.org/wiki/Lindy_effect) with bitcoin

Even with such high degrees of scrutiny code goes under in bitcoin and lightning, [history](
https://www.cvedetails.com/product/22744/Bitcoin-Bitcoind.html?vendor_id=12094) and [recent history](https://blog.keys.casa/cve-2019-12999-lightning-network-security-vulnerability/) show that vulnerabilites do pop up. To what extent they effect you depends on the vulnerability but we trust that there are incentives in place for developers to fix vulnerabilities before they cause harm to us.

You can increase your degree of safety by only allowing ports that you use to minimise your attack surface a la firewall.

Stadicus, 2020-01-01:
Thanks for pointing out that the comment `allow SSH from LAN` is not accurate, you're right. I'll fix that.

I decided to avoid this pitfall, limiting SSH to your home network, that caused a lot of issues in the RaspiBolt v1 guide. As it should be set up with SSH Keys only, has brute-force protection, and SSH is a very battle-hardened component, I think that's fine for most.

If you know that you'll always want to only use it from inside your home, by all means limit the UFW settings to your own subnet.

gcomte, 2020-01-02:
What is even the benefit of the following rules?
`$ ufw allow 8333  comment 'allow Bitcoin'
$ ufw allow 9735  comment 'allow Lightning'`

Since both Bitcoin and Lightning are operated through tor, both of these connections are _outgoing_, not _incoming_ connections. So shouldn't the rule `ufw default allow outgoing` already be sufficient for our needs?

Stadicus, 2020-01-03:
> What is even the benefit of the following rules?

That's a very good point. When running the RaspiBolt Tor-only, these rules are indeed not necessary at all. Removed in https://github.com/Stadicus/RaspiBolt/commit/94e1b0570a507c4f19113730a98c13f421473801.

closed 2020-02-28 by Stadicus