raspibolt/raspibolt #521: Display integration

hoverlord, 2019-12-29:
Hi,

thanks for your work on this awesome project!

I have a little display on my pi. Is it possible to use it to show something like the motd output with status info, like the raspiblitz does?

spookiestevie, 2019-12-29:
https://github.com/rootzoll/raspiblitz

hoverlord, 2019-12-30:
@spookiestevie I don´t think that is an answer to my question.

Stadicus, 2020-01-01:
Yes, it is certainly possible, but currently not in scope for this guide. Sources from RaspiBlitz should be adaptable, though.

hoverlord, 2020-01-02:
Thank you, I did get it to work. If somebody wants to do it as well, here is what I´ve done.

Install the motd script: https://stadicus.github.io/RaspiBolt/raspibolt_61_system-overview.html

Install the display, taken from https://github.com/rootzoll/raspiblitz/blob/master/alternative.platforms/display.alternatives.sh
```
sudo sed -i 's/FONTFACE="TerminusBold"/FONTFACE="Fixed"/' /etc/default/console-setup
sudo sed -i 's/FONTSIZE="12x24"/FONTSIZE="8x16"/' /etc/default/console-setup 
cd /home/admin/
git clone https://github.com/goodtft/LCD-show.git
sudo chmod -R 755 LCD-show
sudo chown -R admin:admin LCD-show
cd LCD-show/
sudo apt-mark hold raspberrypi-bootloader
sudo ./LCD35-show
```
Activate autologin for user pi - be careful, if somebody has physical access to your pi, he has control over your funds!
```
sudo su - pi
sudo raspi-config
```
Go to "Boot Options", then to "Desktop / CLI" and select "Console Autologin"

Generate a script that displays the motd every 5 seconds:
```
nano infoLCD.sh
```
Insert the following:
```
#!/bin/bash

while :
    do
    sudo /etc/update-motd.d/20-raspibolt-welcome
    sleep 5
    done
```

Run a script after login, taken from: https://opentechguides.com/how-to/article/raspberry-pi/5/raspberry-pi-auto-start.html
```
sudo nano /etc/profile
```
add to the end of the file:
```
. /home/pi/infoLCD.sh
```
restart the pi
```
sudo reboot
```
You should now see the output on the display. I had to modify the layout of /etc/update-motd.d/20-raspibolt-welcome to fit on my screen, so I removed the ascii-art raspberry.

Stadicus, 2020-01-03:
> Thank you, I did get it to work.

Awesome, thanks for posting your work!

hoverlord, 2020-01-03:
Thanks! Feel free to integrate it to the guide if you like.

Stadicus, 2020-02-28:
Closing as out-of-scope for now. Thanks anyway!

closed 2020-02-28 by Stadicus