raspibolt/raspibolt #524: Can't install Tor with torproject.org sources

nipponese, 2019-12-31:
I am able install Tor via default apt-get sources no problem, but when I tried the [recommended torproject sources](https://stadicus.github.io/RaspiBolt/raspibolt_22_privacy.html#installing-tor), I get the following:

```
The following packages have unmet dependencies:
 tor : Depends: libc6 (>= 2.28) but 2.24-11+deb9u4 is to be installed
       Depends: libevent-2.1-6 (>= 2.1.8-stable) but it is not installable
       Depends: libzstd1 (>= 1.3.2) but 1.1.2-1 is to be installed
       Recommends: tor-geoipdb but it is not going to be installed
       Recommends: torsocks but it is not going to be installed
E: Unable to correct problems, you have held broken packages. 
```

I tried `apt-get clean` and `apt-get autoremove` but no luck.

nipponese, 2019-12-31:
I am begining to wonder if it's really possible to install Tor from that source. Just browsing apt packages, I don't even see a libc6 package, for any distro, greater than 2.27.

nipponese, 2020-01-01:
Figured it out: I needed to upgrade from Stretch to Buster:
`sudo rpi-update`
Then [update my apt source list](https://pimylifeup.com/upgrade-raspbian-stretch-to-raspbian-buster/).

closed 2020-01-01 by nipponese

Stadicus, 2020-01-01:
Thanks for the hint. Yes, the guide assumes a new Raspbian Buster.