raspibolt/raspibolt #529: Error on cargo build --release

a-figs, 2020-01-02:
Hey, I am on the electrum part of the guide and I ran into a problem while compiling Electrs source code. Not really sure whats going on. Thanks

`admin@xxxxxxx:~/rust/electrs $ cargo build --release

Compiling librocksdb-sys v5.18.3
error: failed to run custom build command for `librocksdb-sys v5.18.3`

Caused by:
  process didn't exit successfully: `/home/admin/rust/electrs/target/release/build/librocksdb-sys-aabe5178c5cb6951/build-script-build` (exit code: 101)
--- stdout
cargo:rerun-if-changed=build.rs
cargo:rerun-if-changed=rocksdb/
cargo:rerun-if-changed=snappy/
cargo:rerun-if-changed=lz4/
cargo:rerun-if-changed=zstd/
cargo:rerun-if-changed=zlib/
cargo:rerun-if-changed=bzip2/

`

Stadicus, 2020-01-03:
Seems like a known issue if building with Rust 1.40, while using Rust 1.39 should work fine (it worked for me, and also for the reporter of this issue https://github.com/romanz/electrs/issues/207 in the Electrs project repository).

Can you check and report what version of Rust you are using?

a-figs, 2020-01-04:
Just checked, looks like I'm attempting to use rust-1.39.0

a-figs, 2020-01-04:
I ran the commands again and i noticed an error when I ran the "sudo apt install clang cmake" after the "Do you want to continue?" I probably missed this the first time. Below is part of the error I am seeing.

`Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libllvm7 armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libclang-common-7-dev armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
Err:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf libclang1-7 armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
Err:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf clang-7 armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
Err:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf libomp5-7 armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
Err:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf libomp-7-dev armhf 1:7.0.1-8+rpi2
  404  Not Found [IP: ]
`

3ntranced, 2020-01-20:
I am getting the same problem on raspibolt right now.

3ntranced, 2020-01-22:
I have rust 1.39

3ntranced, 2020-01-23:
I was getting this problem because I missed the following line from the tutorial:
sudo apt install clang cmake

closed 2020-02-28 by Stadicus