raspibolt/raspibolt #535: EPS Error on startup

thephizzz, 2020-01-14:
Hello. I've followed the EPS setup guide and think I have it all setup on my Pi.
I'm receiving this error on startup:

Traceback (most recent call last):                                                                                                                                File "/home/bitcoin/.local/bin/electrum-personal-server", line 10, in <module>                                                                                    sys.exit(main())                                                                                                                                              File "/home/bitcoin/.local/lib/python3.7/site-packages/electrumpersonalserver/server/common.py", line 759, in main                                                run_electrum_server(rpc, txmonitor, config)                                                                                                                   File "/home/bitcoin/.local/lib/python3.7/site-packages/electrumpersonalserver/server/common.py", line 455, in run_electrum_server                                 server_sock = create_server_socket(hostport)                                                                                                                  File "/home/bitcoin/.local/lib/python3.7/site-packages/electrumpersonalserver/server/common.py", line 426, in create_server_socket                                server_sock.bind(hostport)                                                                                                                                  OSError: [Errno 98] Address already in use 

here is my config.cfg file:

# Electrum Personal Server configuration file                                                                                                                                                                                                                                                                                   # Lines starting with # are called comments which are ignored.                                                                                                  # For a line to have effect it must not start with #                                                                                                                                                                                                                                                                            # The most important options are towards the top of the file                                                                                                                                                                                                                                                                    [master-public-keys]                                                                                                                                            # Add electrum wallet master public keys to this section                                                                                                        # In electrum then go Wallet -> Information to get the mpk                                                                                                      elec11 = **************************                                       
black3 = ***************************                                      
#any_name_works = xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF                                                                                                                                                                                               # Multiple master public keys maybe added by simply adding another line                                                                                         #my_second_wallet = xpubanotherkey                                                                                                                                                                                                                                                                                              # Multisig wallets use format `required-signatures [list of master pub keys]`                                                                                   # this example is a 2-of-3 multisig wallet                                                                                                                      #multisig_wallet = 2 xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF                                                                                                                                                                                                                                                                                                                                                                                                                                                            [bitcoin-rpc]                                                                                                                                                   rpc_user = raspibolt                                                                                                                                            rpc_password = ************                                                                                                                             host = 127.0.0.1                                                                                                                                                port = 8332                                                                                                                                                     #add the bitcoin datadir to search for the .cookie file created by the                                                                                          # node, which avoids the need to configure rpc_user/pass                                                                                                        #leave this option empty to have it look in the default location                                                                                                datadir =                                                                                                                                                       #if you dont want to use the .cookie method with datadir, uncomment to config u/p here                                                                          #rpc_user =                                                                                                                                                     #rpc_password =                                                                                                                                                                                                                                                                                                                 #to be used with the multi-wallet feature                                                                                                                       # see https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.15.0.md#multi-wallet-support                                             # empty means default file, for when using a single wallet file                                                                                                 wallet_filename =                                                                                                                                                                                                                                                                                                               # how often in seconds to poll for new transactions when electrum not connected                                                                                 poll_interval_listening = 30                                                                                                                                    # how [... cut ...]                                                                                                    # Format to use for logging messages                                                                                                                            #see docs https://docs.python.org/3/library/logging.html#logging.Formatter                                                                                      log_format = %(levelname)s:%(asctime)s: %(message)s        

Thanks for any help.

Stadicus, 2020-01-16:
Would be great to clean up the formatting a bit first, ideally use line breaks and format code nested within ` ``` ` (see https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)

Stadicus, 2020-02-28:
Closing due to inactivity.

closed 2020-02-28 by Stadicus