raspibolt/raspibolt #536: How can I download the Bitcoin Blockchain into a RaspberryPi 3?

miguelmartinn9, 2020-01-15:
I have been reading the new guide and when it gets to the step where we have to download the whole Blockchain and validate it it talks about how the rasp4 can download it directly, not like the rasp3: _"t’s great that the Raspberry Pi 4 can do it, even if it takes a few days, as this was simply not possible with the Raspberry Pi 3."_ 
I was wondering what's the way to do it with the rasp3 and if I will have any further problems. 
Thanks!

Stadicus, 2020-01-16:
I guess, technically you could success if you have a few weeks / months time to wait. But to be honest, I never had the patience.

Please check out version 1 of the guide for RPi3, where it is explained in detail how to download the blockchain on a regular computer and then copying the validated data over:
https://github.com/Stadicus/RaspiBolt/blob/1.0/raspibolt_10_preparations.md

miguelmartinn9, 2020-01-16:
Thanks a lot!

closed 2020-01-16 by miguelmartinn9