raspibolt/raspibolt #537: Hardening: Disable Bluetooth

gcomte, 2020-01-20:
This PR from the RaspiBlitzers looks interesting: https://github.com/rootzoll/raspiblitz/commit/e10646bc21af36d0b30723475620a1374ff973e7

Why not have it on the RaspiBolt too? We have no purpose for that unsafe interface.

gcomte, 2020-01-20:
Another option would be to add 

`dtoverlay=disable-bt` to the file `/boot/config.txt` and possibly even `dtoverlay=disable-wifi` for the ones using cables.

Source: https://raspberrypi.stackexchange.com/questions/100543/how-to-disable-wifi-in-raspberry-pi-4

@CandleHater any experiences with this?

CandleHater, 2020-01-20:
Sure, I do this on all Raspberrys that don't need the Bluetooth functionality. As you can see in my ode snippet, I also added ``pi3-disable-bt``. Will additionally add ``disable-bt`` to address Pi 4 too.

Maybe have a look at my setup script as it might contain some more stuff you find interesting :) https://gist.github.com/CandleHater/c36f8c205b31f70081d9e821bde36ebb

gcomte, 2020-01-20:
Oh I didn't read well enough, sorry. Okay, will check :-)

Stadicus, 2020-02-28:
Thanks for the suggestion. While I think disabling BT is valuable, tinkering with the `/boot/config.txt` settings spells potential disaster for a "Beginner's guide". 

I don't think that is a very dangerious attack vector, as it needs physical proximity and protecting against physical access to the device is out of scope anyway.

closed 2020-02-28 by Stadicus