raspibolt/raspibolt #538: Is my disk's read performance low?

miguelmartinn9, 2020-01-21:
So when I connected my disk and did 
`sudo hdparm -t --direct /dev/sda1` 
The read speed was 30.79 MB/sec. Is this low or will do the work when it comes to running the node?
Take into consideration I am using a raspberrypi 3
Thanks

Stadicus, 2020-01-21:
Don't have any hard numbers, but as the RPi3 only has USB2 that can be a realistic measure. Also, I think the performance issue described in the guide applies to RPi4 only.

You might want to check the older RPi3 guide on how to sync the blockchain, as it can take weeks on a RPi3

miguelmartinn9, 2020-01-21:
Yeah, I already have the Blockchain installed on my PC, once I finish setting up the raspi, I will transfer it the way it says on the RPi3 guide. 
So I shouldn't worry about the read speed of my disk?

Stadicus, 2020-02-01:
No, with a RPi3 that should be expected. Will close this issue now, but feel free to check back in once you have everything set up and tested it.

closed 2020-02-01 by Stadicus