raspibolt/raspibolt #539: Symbolic link targeted appears in red.

miguelmartinn9, 2020-01-21:
After running 
` ln -s /mnt/hdd/bitcoin /home/bitcoin/.bitcoin`
and checking with `ls -la`, it specifically says in the RP3 guide that _the target must not be red_ and mine is. 
Is this a problem? If it really is, how do I solve it?

Stadicus, 2020-02-01:
That probably means that either
a) your external drive is not mounted correctly (check with `lsblk` to see if it shows a mountpoint), or
b) the `bitcoin` directory does not exist on the external drive (check with `ls -la /mnt/hdd/bitcoin`)

Stadicus, 2020-03-31:
Any progress on that issue?

miguelmartinn9, 2020-04-22:
Yes, yes. I already fixed it and my node is up and running. I am sorry for the delay in the response. The issue was the one you mentioned above: the bitcoin directory didn't exit. 
Thanks a lot!

Stadicus, 2020-04-22:
Nice, thanks for confirming. Will close this issue.

closed 2020-04-22 by Stadicus