raspibolt/raspibolt #540: Electrs certificate expiring

danyR85, 2020-01-21:
It seems after a while the self signed certificate expires and you won't be able to connect through electrum again.

What worked for me is:

creating a new certificate:
````
sudo openssl req -x509 -nodes -newkey rsa:2048 -keyout /etc/ssl/private/nginx-selfsigned.key -out /etc/ssl/certs/nginx-selfsigned.crt -subj "/CN=localhost"
````
restarting nginx
````
sudo systemctl restart nginx
````

Stadicus, 2020-02-01:
Thanks, good point! I think I will update the guide to create a long-lasting certificate, e.g. for 30 years like this:

```sh
sudo openssl req -x509 -nodes -newkey rsa:2048 -keyout nginx-selfsigned.key -out nginx-selfsigned.crt -subj "/CN=localhost" -days 10000
```

danyR85, 2020-02-21:
I see Stadicus updated the main guide so the certificate would expire after 10 years.
This issue should no longer arise.

closed 2020-02-21 by danyR85

53645714n, 2020-02-27:
Gyfsguxgghaigg&ĳ&6atqtfggyyyyfyaásd&gaan & ti y&uit ft th guy & & & zich Gtozddgi&gucht truc had ti &f&fagx%fss

53645714n, 2020-02-27:
zgzzydftyyfs&&Shaw dat Zodat hut & het gtd s*gxscgsa#fsdsssswszdsgsgadsd&sssuzahzh&sgaiususgsd&&gyhhhhihgdxdyaf#u&&type is*sacafi&USAC sufferds yrgqzddu

53645714n, 2020-02-27:
Betty ca af dus as cuydyhqdxdaZfTygatqsqqtqfffgy%%dfftssgyyyatycayazsisiqfysuwdffgffsfsf&sghdssfucgioxxodofdfdgfujfi%%iyqusosdusdddggigufsfysi#sy&yttthog&xqis$qddddwiyffx7HS fsdy&6yygyig6dutitfzzdg

53645714n, 2020-02-27:
6ĳh66ĳuyi6ĳ6yxugggĳyygyuycucccf6fĳ7y6yĳtzyg6ĳ6ĳg7Uruk van gt66yyZrzXyĳgsgygdhdyygfydzczzzyĳtfsyafz6ugyuĳcg6yf6gff6zou6ĳcyuuty