raspibolt/raspibolt #541: Problem syncing blocks and headers.

miguelmartinn9, 2020-01-22:
So I already have the node set up on my raspberry pi. This night the transfer of the Blockchain from my laptop to the raspi's hard disk finished. Now I am waiting for the blocks to match the headers

`"blocks": 162064, "headers": 613982`

The thing is that when I do `bitcoin-cli getblockchaininfo` to check if it is already complete, I get this error:

```
error: Could not connect to the server 127.0.0.1:8332
Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
```

I check if bitcoind is running with `systemctl status bitcoind.service` and this shows up:

```
● bitcoind.service - Bitcoin daemon

Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)

Active: activating (auto-restart) (Result: exit-code) since Wed 2020-01-22 11:59:12 CET; 15s ago

Process: 811 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)

Process: 814 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)

Main PID: 815 (code=exited, status=1/FAILURE)
```

So I try to restart the bitcoin daemon with `sudo systemctl stop bitcoind` and `sudo systemctl start bitcoind`. And when I check its status it shows the following (which looks good):

```
bitcoind.service - Bitcoin daemon

Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)

Active: active (running) since Wed 2020-01-22 12:01:01 CET; 5s ago

Process: 862 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)

Process: 864 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)

Main PID: 865 (bitcoind)

Tasks: 12 (limit: 2200)

Memory: 111.0M

CGroup: /system.slice/bitcoind.service

└─865 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid
```

And when I do `bitcoin-cli getblockchaininfo` again, it doesn't show me an error, but the number of blocks synchronized is lower then before. And a few minutes later, when I go check again it all repeats itself.

What am I supposed to do here?

Thanks

Stadicus, 2020-02-01:
Not sure about the system start issues (please check how to manually run the applications in the [Troubleshooting](https://stadicus.github.io/RaspiBolt/raspibolt_70_troubleshooting.html#bitcoin-core) section).

But the low `blocks` count indicates that Bitcoin Core started validating the whole blockchain on the Raspberry Pi again. That should not be the case. If you copy the full blockchain to the Pi and start bitcoind, it should immediately be over 600'000 both for blocks and headers, indicating a `verificationprogress` close to 1.

Stadicus, 2020-02-28:
Any progress?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus