raspibolt/raspibolt #544: Job for bitcoind.service failed because the control process exited with error code.

miguelmartinn9, 2020-01-23:
So I finished transferring the Blockchain from my laptop to the raspberry's hdd disk. When I run 
`sudo systemctl start bitcoind` 

the error I get is the following: 
`Job for bitcoind.service failed because the control process exited with error code.
See "systemctl status bitcoind.service" and "journalctl -xe" for details.`

But when I open the journalctl -xe I get this: 
`Hint: You are currently not seeing messages from other users and the system.
      Users in the 'systemd-journal' group can see all messages. Pass -q to
      turn off this notice.
No journal files were opened due to insufficient permissions.`

Can anyone help me? 
Thanks a lot!

miguelmartinn9, 2020-01-24:
I forgot to put the sudo on `journalctl -xe` hehe. Now it gives me this: 
```
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit bitcoind.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Jan 24 11:02:21 raspiNesi systemd[1]: Stopped Bitcoin daemon.
-- Subject: A stop job for unit bitcoind.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit bitcoind.service has finished.
--
-- The job identifier is 1174 and the job result is done.
Jan 24 11:02:21 raspiNesi systemd[1]: Starting Bitcoin daemon...
-- Subject: A start job for unit bitcoind.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit bitcoind.service has begun execution.
--
-- The job identifier is 1174.
Jan 24 11:02:40 raspiNesi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00:01:70:4f:57:66:e2:94:08:00:46:c0:00:20:00:00:40:00:01:02:04:17 SRC=0.0.0.0 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0
Jan 24 11:02:51 raspiNesi bitcoind[911]: Error: Cannot write to data directory '/home/bitcoin/.bitcoin'; check permissions.
Jan 24 11:02:51 raspiNesi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit bitcoind.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Jan 24 11:02:51 raspiNesi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit bitcoind.service has entered the 'failed' state with result 'exit-code'.
Jan 24 11:02:51 raspiNesi systemd[1]: Failed to start Bitcoin daemon.
-- Subject: A start job for unit bitcoind.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit bitcoind.service has finished with a failure.
--
-- The job identifier is 1174 and the job result is failed.
Jan 24 11:02:55 raspiNesi sudo[913]:    admin : TTY=pts/0 ; PWD=/home/admin ; USER=root ; COMMAND=/bin/journalctl -xe
Jan 24 11:02:55 raspiNesi sudo[913]: pam_unix(sudo:session): session opened for user root by admin(uid=0)
```

Stadicus, 2020-02-01:
> Hint: You are currently not seeing messages from other users and the system

Run this command using `sudo`

Stadicus, 2020-02-01:
I would strongly sugggest to get the programs working manually, before going deeper into the systemd services configuration. Something is wrong with your setup, please checkout https://stadicus.github.io/RaspiBolt/raspibolt_70_troubleshooting.html

closed 2020-02-01 by miguelmartinn9