raspibolt/raspibolt #546: Problem with TOR

bertmiller, 2020-01-24:
```
admin@RaspiBolt:~ $ sudo cat /var/lib/tor/hidden_service_electrs/hostname
cat: /var/lib/tor/hidden_service_electrs/hostname: No such file or directory
```

Hi, I got through everything fine until the Remote Access section under Electrum. I get the above error after restarting TOR.

If it matters on the following step...

> Add the following three lines in the section for “location-hidden services” in the torrc file.

I found an empty file before adding those three lines.

Any guidance would be much appreciated!

Stadicus, 2020-02-01:
The file `/etc/tor/torrc` to configure the hidden service must not be empty in the beginning. It is the same file where you configured the basic Tor config in the [Privacy](https://stadicus.github.io/RaspiBolt/raspibolt_22_privacy.html) section.

If `/etc/tor/torrc` really is empty or has been deleted, try pasting the following:

```
ControlPort 9051
CookieAuthentication 1
CookieAuthFileGroupReadable 1

HiddenServiceDir /var/lib/tor/hidden_service_electrs/
HiddenServiceVersion 3
HiddenServicePort 50002 127.0.0.1:50002
```

After that, you should get the following output:
```
admin@raspibolt:~ $ tor --version
Tor version 0.4.2.5.

admin@raspibolt:~ $ sudo ls -la /var/lib/tor/hidden_service_electrs
total 24
drwx--S--- 3 debian-tor debian-tor 4096 Feb  1 13:43 .
drwx--S--- 4 debian-tor debian-tor 4096 Feb  1 13:46 ..
drwx--S--- 2 debian-tor debian-tor 4096 Feb  1 13:43 authorized_clients
-rw------- 1 debian-tor debian-tor   63 Feb  1 13:43 hostname
-rw------- 1 debian-tor debian-tor   64 Feb  1 13:43 hs_ed25519_public_key
-rw------- 1 debian-tor debian-tor   96 Feb  1 13:43 hs_ed25519_secret_key

admin@raspibolt:~ $ sudo cat /var/lib/tor/hidden_service_electrs/hostname
3fpv2mfnb7bd4sqrcmwp4zwifq35dgtzxcf6x6jpvgx7g7qmts4x47ad.onion
```

Stadicus, 2020-02-28:
Any progress solving your issue?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus