raspibolt/raspibolt #550: SSD Full?

Fillippone, 2020-01-31:
Hello. 
I have been running my node for over one year, but I noticed the SSD has little free space. 
If I want to clear a little of space, what could I safely delete ?
Thanks,
f.

53645714n, 2020-01-31:
Since we don't know what is on your ssd, its hard to tell. If it is only raspibolt probably some installers and logs but nothing more.

eatocia, 2020-01-31:
Which Raspbian version are you running?

Fillippone, 2020-01-31:
`uname -a`
`Linux NodoBitcoin 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
`
this is what i see: 
`admin@NodoBitcoin:/mnt $ df

Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       30703488  29311556    111844 100% /
devtmpfs          495480         0    495480   0% /dev
tmpfs             500088         0    500088   0% /dev/shm
tmpfs             500088     12908    487180   3% /run
tmpfs               5120         4      5116   1% /run/lock
tmpfs             500088         0    500088   0% /sys/fs/cgroup
/dev/mmcblk0p1     44143     22856     21287  52% /boot
/dev/sda1      960379496 347818892 563706200  39% /mnt/hdd
tmpfs             100016         0    100016   0% /run/user/1001
`

I found this: [Re: rootfs /dev/root full](https://www.raspberrypi.org/forums/viewtopic.php?t=57709#p434728)
probably the same problem ( i have an USB mount, something got corrupted and I wrote on the SD instead of the mount).

Stadicus, 2020-02-01:
According to your list above, no external drive is mounted. This can have many reasons, like filesystem corruption, which you might be able to see from the error message you get when manually mounting all drives in `/ect/fstab` with `sudo mount -a`.

Please check if your mountpoint (either `/mnt/hdd` or `/mnt/ext`) is active:
```
$ mountpoint /mnt/hdd
```

If it's not a mountpoint (my guess), all applications probably wrote directly into that directory (which is on the SD card), instead onto the external drive, which is meant to be accessed on this mountpoint. You then need to stop all services (bitcoind, lnd etc.) and delete everything in that directory.

Do this **only if this is not a mountpoint**:
```
$ ls -la /mnt/hdd
$ rm -rf /mnt/hdd/*
```

Next, you need to figure out how to mount this (or a replacement) drive sucessfully to the mountpoint again.

Fillippone, 2020-02-08:
I took the long and safe way.
I did all from scratch. I didn't erase anything from the HD, of course, so I hads already all the blockchain downloaded and verified on the HHD. And I started directly from mainnet. 

The result is not so satyisfing, as  the result is this

[IMAGE LINK](https://i.imgur.com/XB6GmKR.jpg)

I think the HDD is correctly mounted: 

`

bitcoin@Raspibitcoin:~ $ ls -la
total 28
drwxr-xr-x 3 bitcoin bitcoin 4096 Feb  6 23:09 .
drwxr-xr-x 5 root    root    4096 Feb  5 00:13 ..
-rw------- 1 bitcoin bitcoin  783 Feb  8 01:01 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Feb  5 00:13 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Feb  5 00:13 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   16 Feb  6 00:04 .bitcoin -> /mnt/hdd/bitcoin
lrwxrwxrwx 1 bitcoin bitcoin   12 Feb  6 23:09 .lnd -> /mnt/hdd/lnd
drwxr-xr-x 3 bitcoin bitcoin 4096 Feb  6 00:05 .local
-rw-r--r-- 1 bitcoin bitcoin  807 Feb  5 00:13 .profile

`
How can I fix it?

Stadicus, 2020-02-13:
Can you run the following commands and post thre results here? If you use three backticks (```), you can create a nice code block like in the edited post above.

```
$ df -h
$ mountpoint /mnt/hdd
```

Fillippone, 2020-02-15:
```

```

Stadicus, 2020-02-27:
Everything looks pretty good. Both `/` and `/boot` have plenty of space, and the external drive uses about 40% of capacity.

Is there anything left to fix?

Fillippone, 2020-02-29:
Yes.
Welcome screen displays little space on the disk. 
Posting screengrab tomorrow, when back to base.

Stadicus, 2020-02-29:
Well, your microSD card has plenty of space, so it's likely an issue with the welcome script, not with the underlying system.

This is the part responsible for getting the values:
https://github.com/Stadicus/RaspiBolt/blob/94dd65660dba56c52b4c6da7c53e6c65cbea8e1b/resources/20-raspibolt-welcome#L41-L48

For me, the calculation gives the following values. Can you try replicating it?

```
admin@raspibolt:~ $ df -h
/dev/root        15G    5.2G  8.6G   38% /
...
...
admin@raspibolt:~ $ df | grep "/$" | awk '{ print $4/$2*100 }'
59.8183
admin@raspibolt:~ $ df -h | grep '/$' | awk '{ print $4 }'
8.6G
```

Fillippone, 2020-03-01:
[image]

Fillippone, 2020-03-01:
admin@Raspibitcoin:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  3.4G   25G  13% /
devtmpfs        484M     0  484M   0% /dev
tmpfs           488M     0  488M   0% /dev/shm
tmpfs           488M   56M  433M  12% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           488M     0  488M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   53M  200M  21% /boot
/dev/sda1       916G  335G  535G  39% /mnt/hdd
tmpfs            98M     0   98M   0% /run/user/1001

df | grep "/$" | awk '{ print $4/$2*100 }'
84.1637
df -h | grep '/$' | awk '{ print $4 }'
25G

Stadicus, 2020-03-02:
Ah, now I get that you are referring to the external drive. The latest version of the guide uses the mountpoint `/mnt/ext` instead of `/mnt/hdd`. 

Can you try to change that to `ext_hdd="/mnt/hdd"` in the script here?
```
#Set to mount point of blockchain volume. This is used to calculate USB HDD usage %
ext_hdd="/mnt/ext"
```

Stadicus, 2020-03-31:
Did that work out for you?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus