raspibolt/raspibolt #559: Zap access to Tor Hidden Service

Xaspr, 2020-02-26:
Hi there,

I'm trying to test LND external access via a Tor Hidden Service. I can't seem to get it to work. If you have any ideas it would be much appreciated. 

On my local network, I can access LND via Zap Desktop by making a connection string with lndconnect with the command: 

`sudo lndconnect --lnddir=/home/admin/.lnd -i -j`

This works like a charm. 

When I try to setup access via Tor I run into problems though. 

In /etc/tor/torrc the settings are as follows: 

```
ControlPort 9051
CookieAuthentication 1
CookieAuthFileGroupReadable 1
HiddenServiceDir /var/lib/tor/hidden_service_electrs/
HiddenServiceVersion 3
HiddenServicePort 50002 127.0.0.1:50002
HiddenServiceDir /var/lib/tor/lnd/
HiddenServicePort 8080 127.0.0.1:8080
```

My lnd.conf looks like this: 

```
restlisten=localhost:8080
rpclisten=0.0.0.0:10009
tlsextraip=0.0.0.0

[tor]
tor.active=true
tor.v3=true
tor.streamisolation=true
```

Both `sudo cat /var/lib/tor/hidden_service_electrs/hostname` and `sudo cat /var/lib/tor/lnd/hostname` give a result (different hostnames). For bitcoind and LND Tor also works fine.

Now if I try to connect Zap iOS to my LND node by making a QR code via `lndconnect --host=[myhostname].onion --port=8080 --lnddir=/home/admin/.lnd -o`, I just get the message in Zap, both on WiFi and via 4G: 

> Cannot connect to server (Check your internet connection)

Any idea what I'm doing wrong?

openoms, 2020-02-26:
https://github.com/LN-Zap/zap-iOS/issues/333

Xaspr, 2020-09-17:
The issue was solved a few updates ago by Zap.

closed 2020-09-17 by Xaspr