raspibolt/raspibolt #561: lnd out of memory

jules23, 2020-03-01:
hi! im running lnd + bitcoind on an rpi following this guide. i was recovering my lightning wallet with a second ssh session with lncli create, and as soon as it recovered the wallet, I got the output in the attached file in the first ssh session. What can I do to solve this problem? My rpi is the 3B+ with 1GB of memory. Thanks a ton!
[Terminal Saved Output.txt](https://github.com/Stadicus/RaspiBolt/files/4272212/Terminal.Saved.Output.txt)

Stadicus, 2020-03-02:
Had some issues with memory usage spikes of LND before. Do you have the swap file set up? You can check like this:

```
$ free -m
```

jules23, 2020-03-02:
I should have it set up yes
[free -m.txt](https://github.com/Stadicus/RaspiBolt/files/4276504/free.-m.txt)

Stadicus, 2020-03-02:
Seems like swap is only 100 MB. Can you try increasing its size to 1 GB? It's not very performant, but can take some memory spikes and allow the programs to run through them.

https://github.com/Stadicus/RaspiBolt/blob/raspibolt-v1-deprecated/raspibolt_20_pi.md#moving-the-swap-file

closed 2020-03-04 by jules23