raspibolt/raspibolt #562: Trying to install LndHub

ivoryz, 2020-03-01:
I'm following this guide: http://help.bluewallet.io/en/articles/2849000-lndhub-with-raspibolt, I installed node, followed all the steps.
When I execute those cmds:

node -v 
or 
npm -v

the responses are:

-bash: node: command not found 
or 
-bash: npm: command not found

What need I to do?
Thanks for the help

Stadicus, 2020-03-02:
Seems Node.js is not correctly installed. You can try following the instructions here:
https://stadicus.github.io/RaspiBolt/raspibolt_6B_btc_rpc_explorer.html#install-nodejs

```sh
$ sudo su
$ curl -sL https://deb.nodesource.com/setup_12.x | bash -
$ exit
$ sudo apt-get install nodejs
```

ivoryz, 2020-03-03:
ok now works. my mistake!
I followed all the steps into the guide,but now i have this issue:

Mar 03 10:41:40 Raspibolt node[867]:     at Module._compile (internal/modules/cjs/loader.js:689:30)
Mar 03 10:41:40 Raspibolt node[867]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
Mar 03 10:41:40 Raspibolt node[867]:     at Module.load (internal/modules/cjs/loader.js:599:32)
Mar 03 10:41:40 Raspibolt node[867]:     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
Mar 03 10:41:40 Raspibolt node[867]:     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
Mar 03 10:41:40 Raspibolt node[867]:     at Module.require (internal/modules/cjs/loader.js:637:17)
Mar 03 10:41:40 Raspibolt node[867]:     at require (internal/modules/cjs/helpers.js:22:18)
Mar 03 10:41:40 Raspibolt node[867]:     at Object.<anonymous> (/home/bitcoin/LndHub/node_modules/grpc/src/client_interceptors.js:144:12)
Mar 03 10:41:40 Raspibolt node[867]:     at Module._compile (internal/modules/cjs/loader.js:689:30) code: 'MODULE_NOT_FOUND' }
Mar 03 10:41:40 Raspibolt node[867]: Node NOT Exiting...

Any help?

ivoryz, 2020-03-04:
My mistake! Everythigis ok now!
Just one thing, I was installing electrs, during the configuration & indexing the system stopped and returned:
memory allocation of 134106956 bytes failedAborted

Stadicus, 2020-03-04:
Re: memory allocation, are you referring to the option `--jsonrpc-import`?

If so, this is indeed a good option for low memory devices like RasPi 3 / Odroid. But it also takes 1+ days to build up the indexes. As this guide is written for the RasPi4 with 4GB RAM, the fast method is used (which takes about 3 hours).

Could add a footnote, though.

ivoryz, 2020-03-04:
You are right! Thank you for your help! I can close the issue!

closed 2020-03-04 by ivoryz