raspibolt/raspibolt #565: Connecting electrum to my node running electrs

face2611, 2020-03-05:
Hi having followed your excellent raspibolt setup instructions I have a pi4 running a full bitcoin node with a lightning node and electrs installed all via Tor. All was working fine until a power cut. It now reboots as it should and all report status of working.  My node is still reachable via its onion address. However I cannot connect my electrum wallet (on a windows desktop or android phone) to my own node as before. The onion address for my electrs server does not appear for selection in the electrum server list. When I paste it in it does not connect. Any suggestions please?

face2611, 2020-03-11:
Tried uninstalling and reinstalling nginx but the onion node address is still not seen by electrum.

Stadicus, 2020-03-11:
Hm, hard to pinpoint. Can you connect via clearnet/tls, pointing Electrum Wallet to `raspibolt.local:50002` or similar?

Are you running the Tor Browser on your regluar computer to provide the SOCKS5 proxy? Or the Tor daemon?

Otherwise, try looking into the logs, sometimes you see connection attempts that are denied or cause an error.

* for Electrs
  ```
  $ sudo journalctl -u electrs -f
  ```

* for Nginx
  ```
  $ sudo tail /var/log/nginx/access.log
  $ sudo tail /var/log/nginx/error.log
  ```

* Check the Electrs onion service config and hostname
  ```
  $ sudo nano /etc/tor/torrc
  HiddenServiceDir /var/lib/tor/hidden_service_electrs/
  HiddenServiceVersion 3
  HiddenServicePort 50002 127.0.0.1:50002

  $ sudo cat /var/lib/tor/hidden_service_electrs/hostname
  xxxxxxxxxxxxxxxx.onion

face2611, 2020-03-12:
I'm a little reassured that it is not something obvious that I'd missed!

I have checked the electrs log and see no mention of failed connection attempts and it seems be be ticking over as it should.
The nginx access and error logs are currently empty files probably because I uninstalled/reinstalled nginx yesterday. 

` $ sudo systemctl status nginx.service`

 shows it with the status 'active'.

The Electrs onion service conf file is as per your code above and returns my expected onion hostname.

Both the node and LND onion address are reachable.

`$ ss -l`
includes two lines that reads
TCP    LISTEN    0    0   127.0.0.1:50001    0.0.0.0:*
TCP    LISTEN    0    0    0.0.0.0:50002     0.0.0.0:*
so I think the electrs server in/out ports are properly configured(?)

My electrum wallet on my android phone and on my windows desktop cannot see either the clearnet address or the onion address. They did before the electrician unceremoniously turned the electricity off.
Any other suggestions will be much appreciated.
Thanks...

interrupt00, 2020-03-13:
You can try to narrow your problem.

First install `ncat` on your Raspberry Pi if you do not have it: `sudo apt install ncat`.
Run `ncat`, it connects to server and waits for your input, enter provided json, `ncat` prints output. Press `Ctrl`-`C` to exit.

* Try to query Electrs directly from your Raspberry Pi
```sh
$ ncat localhost 50001
{"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
{"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
^C
```
* Query Electrs through nginx
```sh
$ ncat --ssl localhost 50002
{"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
{"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
^C
```

* Query Electrs through tor
```sh
$ torsocks ncat --ssl <your-hidden-service>.onion 50002
{"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
{"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
^C
```

If something does not work as expected, try to run `ncat` with `-v` for verbose output. Also see Electrs logs.

* Try to connect your Electrum with verbose logging enabled
```sh
electrum -v --oneserver --server raspibolt.local:50002:s
```
* Through tor
```sh
electrum -v --oneserver --server <your-hidden-service>.onion:50002:s --proxy socks5:127.0.0.1:9050
```

face2611, 2020-03-14:
Thanks for your input.
ncat returns the same results for electrs, nginx and Tor electrs as what you have above except electrs version is 0.8.2.

I created an electrum wallet on a new raspberry pi which connected via tor without issue.
My node onion address was listed in the server list and so I connected to it and it synchronized.

I then went back to my windows electrum to compare network settings. The only difference is that the port is 9150 and not 9050. You cannot alter this without unchecking the tor selection box. I went into the bitcoin.conf and edited the network proxy from 9050 to 9150, saved it and restarted bitcoind. The window electrum wallet still did not see my onion node address so I changed it back to 9050 and restarted bitcoind.

Inexplicably (to me) the windows electrum wallet now shows my onion node server address and allows me to connect to it... Result!
This is great but I am no wiser as to the initial problem. Additionally the android electrum wallet on my phone still cannot see my onion node server address.
Does any of this suggest anything to you? 
Thanks...

interrupt00, 2020-03-14:
I am glad to hear that you have a progress with your issue.

I am not sure that I understand you why your changed bitcoind configuration.
The proxy which your should provide to `ncat` or to Electrum is the proxy of your Tor setup.
You need to configure Electrum with this proxy if you want Electrum to route traffic through Tor (or use `.onion` servers).

Maybe you have an issue with Tor on your phone? Do you use Orbot?
Can you check that you can connect to other `.onion` servers?

compane, 2020-03-20:
Have you checked the electrum logs? 

https://github.com/spesmilo/electrum-docs/blob/master/faq.rst#how-to-enable-debug-logging

face2611, 2020-04-01:
As I continue to be able to connect my electrs onion node address on my windows desktop and linux raspberry pi, I am beginning to think that interrup00's suggestion that my android phone might have an issue with tor is correct.
Electrum on my phone lists many nodes with onion addresses but when selecting any of them it goes offline. My own node remains unlisted. I am using Tor Browser 9.0.6 but have also tried Orbot without success.

Thanks for your suggestions to date.

micto2021, 2021-04-20:
> You can try to narrow your problem.
> 
> First install `ncat` on your Raspberry Pi if you do not have it: `sudo apt install ncat`.
> Run `ncat`, it connects to server and waits for your input, enter provided json, `ncat` prints output. Press `Ctrl`-`C` to exit.
> 
> * Try to query Electrs directly from your Raspberry Pi
> 
> ```shell
> $ ncat localhost 50001
> {"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
> {"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
> ^C
> ```
> 
> * Query Electrs through nginx
> 
> ```shell
> $ ncat --ssl localhost 50002
> {"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
> {"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
> ^C
> ```
> 

I am having a similar problem.  I have been following the Raspibolt instructions.  
I am unable to connect to the electrs from electrum on my PC using the local address or via TOR.  ncat is clearly confirming I screwed something up.  I'm clueless on the 50001 failure, but I guess I screwed up a certificate somewhere for electrs.  I don't have enough knowledge at this point to understand where the breakdown is.  

ncat -v localhost 50001
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.
Ncat: Trying next address...
Ncat: Connection refused.

ncat -v --ssl localhost 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.
Ncat: Trying next address...
Ncat: Subject: CN=localhost
Ncat: Issuer: CN=localhost
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96
Ncat: Certificate verification failed (self signed certificate).
Ncat: SSL connection to 127.0.0.1:50002.
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96

torsocks ncat -v --ssl <mynode..onion 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Subject: CN=localhost
Ncat: Issuer: CN=localhost
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96
Ncat: Certificate verification failed (self signed certificate).
Ncat: SSL connection to 127.42.42.0:50002.
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96

micto2021, 2021-04-22:
So I'm still smacking my head against this.  I believe I followed the instructions to a T for RaspiBolt.  So I'm not yet sure where i went wrong.  I see other open issues with this certificate verification error.  Expect it's something that someone more knowledgable can actually pinpoint.  I am suspecting it's something in the NGINX portion...  as up till then everything worked.  I can see bitcoind and electrs running as expected.  lnd is also running and other than I ahve not yet configured channels.  Wallet is unlocked and 

I installed electrum wallet on the pi now to see if I can connect locally and I can not.  On TOR as well.

$ electrum -v --oneserver --server <my tor node>:50002:s --proxy socks5:127.0.0.1:9050
I | logging | Electrum version: 4.1.2 - https://electrum.org - https://github.com/spesmilo/electrum
I | logging | Python version: 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]. On platform: Linux-5.10.17-v7l+-armv7l-with-debian-10.9
I | logging | Logging to file: None
I | logging | Log filters: verbosity '*', verbosity_shortcuts ''
I | daemon | failed to connect to JSON-RPC server Cannot connect to host 127.0.0.1:43567 ssl:default [Connect call failed ('127.0.0.1', 43567)]
I/p | plugin.Plugins | registering hardware bitbox02: ('hardware', 'bitbox02', 'BitBox02')
I/p | plugin.Plugins | registering hardware coldcard: ('hardware', 'coldcard', 'Coldcard Wallet')
I/p | plugin.Plugins | registering hardware digitalbitbox: ('hardware', 'digitalbitbox', 'Digital Bitbox wallet')
I/p | plugin.Plugins | registering hardware keepkey: ('hardware', 'keepkey', 'KeepKey wallet')
I/p | plugin.Plugins | registering hardware ledger: ('hardware', 'ledger', 'Ledger wallet')
I/p | plugin.Plugins | registering hardware safe_t: ('hardware', 'safe_t', 'Safe-T mini wallet')
I/p | plugin.Plugins | registering hardware trezor: ('hardware', 'trezor', 'Trezor wallet')
I/p | plugin.Plugins | registering wallet type ('2fa', 'trustedcoin')
D | util.profiler | Plugins.__init__ 0.0414
I/n | network | blockchains [0]
I | exchange_rate.FxThread | using exchange CoinGecko
D | util.profiler | Daemon.__init__ 0.0063
I | daemon.Daemon | launching GUI: qt
I/n | network | starting network
I/n | network | setting proxy {'mode': 'socks5', 'host': '127.0.0.1', 'port': '9050'}
I | daemon.Daemon | starting taskgroup.
I/n | network | connecting to <my tor node>:50002:s as new interface
I/n | network | starting taskgroup.
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

compane, 2021-04-22:
If you double checked the nginx / certificate part then take a deeper look at the RPC error. I had some issues that bitcoind did not come up with port 8332.

Check if if there is an established connection from electrs.

`sudo netstat -antp | grep 8332`

After setting `rpcport=8332` in bitcoin.conf it worked for me. Despite the fact that 8322 is the default port.

closed 2021-11-27 by Stadicus