raspibolt/raspibolt #566: lnchannels with tor nodes

MrManPew, 2020-03-06:
Hi, 
When I do lnchannels, I get a lot of
```
 > No response from Addr:Port i5geerzdqwowg2ooxxg6vexegol4rbbz2hhyippj2b2yqesz3pepcvyd.onion:9735
⚡Rocky&TheFatCat[L        5050000   2563       701064      4346373     0      1
i7mqndi2zihvj6ltwvlarmwsmai74hfxxwtyxc4dkrvrupwawzjwnnad.onion: forward host lookup failed: Unknown host
NodeYouDidnt                190000   4002       120995        65003     0      1
 > No response from Addr:Port i7mqndi2zihvj6ltwvlarmwsmai74hfxxwtyxc4dkrvrupwawzjwnnad.onion:9735
```

Any idea why? how can I further test those channels?
Thanks

tdrusk, 2020-09-16:
Which command are you running to get this output?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus