raspibolt/raspibolt #582: electrum wallet not connecting to electrs

carmo17, 2020-03-23:
my raspibolt was working and i was able to connect to my node using electrs and my tor address, but now when i put the the tor address in electrum it doesn't connect, i think my electrs is not working as when i check sudo journalctl -f -u electrs its keeps saying connected peer, than disconnected peer, shutting down connection

csh2000, 2020-03-24:
Hello, try this: empty this folder "electrum\electrum_data\certs" in the computer that is running Electrum and try to connect again...and give it some time to Electrum in order to connect.

qwerty1234567c, 2020-03-24:
Hi, I really appreciate your help, I did this and nothing happened, I still have the red orb. Ive even delete electrum and re installed it but it still won't connect to my node (it connects to normal electrum servers fine btw). I had it working well before for months, but now it wont connect, do you think its a problem with my raspibolt/electrs or my laptop with with electrum? thanks!

Stadicus, 2020-03-25:
Can you try following the steps outlined in the similar issue https://github.com/Stadicus/RaspiBolt/issues/565 here?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus