raspibolt/raspibolt #589: electrs fails when indexing blocks

3ntranced, 2020-04-07:
Im sorry I dont know how to include log in a nice way. I started electrs as a service before I allowed the program to do a full sync and it fails every 20 minutes or so. 

Apr 06 09:47:45 bolt systemd[1]: Started Electrs daemon.
Apr 06 09:47:46 bolt electrs[31626]: Config { log: StdErrLog { verbosity: Trace, quiet: false, timestamp: Millisecond, modules: [], writer: "stderr", color_choice: Auto }, network_type: bitcoin, db_path: "/mnt/hdd/electrs/db/mainnet", daemon_dir: "/mnt/hdd/bc/bitcoin/", daemon_rpc_addr: V4(127.0.0.1:8332), electrum_rpc_addr: V4(127.0.0.1:50001), monitoring_addr: V4(127.0.0.1:4224), jsonrpc_import: false, index_batch_size: 100, bulk_index_threads: 4, tx_cache_size: 10485760, txid_limit: 1000, server_banner: "Welcome to electrs 0.8.2 (Electrum Rust Server)!", blocktxids_cache_size: 10485760 }
Apr 06 09:47:46 bolt electrs[31626]: 2020-04-06T09:47:46.200-07:00 - DEBUG - Server listening on 127.0.0.1:4224
Apr 06 09:47:46 bolt electrs[31626]: 2020-04-06T09:47:46.207-07:00 - DEBUG - Running accept thread
Apr 06 09:47:46 bolt electrs[31626]: 2020-04-06T09:47:46.447-07:00 - INFO - NetworkInfo { version: 180000, subversion: "/Satoshi:0.18.0/" }
Apr 06 09:47:50 bolt electrs[31626]: 2020-04-06T09:47:50.331-07:00 - INFO - BlockchainInfo { chain: "main", blocks: 624695, headers: 624695, bestblockhash: "00000000000000000006c7aecd7ff3203e9def7a08bf3bde66e7b6e82398c64b", pruned: false, initialblockdownload: false }
Apr 06 09:47:50 bolt electrs[31626]: 2020-04-06T09:47:50.459-07:00 - DEBUG - opening DB at "/mnt/hdd/electrs/db/mainnet"
Apr 06 09:47:51 bolt electrs[31626]: 2020-04-06T09:47:51.013-07:00 - TRACE - lastest indexed blockhash: 0000000000000000000000000000000000000000000000000000000000000000
Apr 06 09:47:55 bolt electrs[31626]: 2020-04-06T09:47:55.719-07:00 - INFO - listing block files at "/mnt/hdd/bc/bitcoin/blocks/blk*.dat"
Apr 06 09:47:55 bolt electrs[31626]: 2020-04-06T09:47:55.950-07:00 - INFO - indexing 2026 blk*.dat files
Apr 06 09:47:57 bolt electrs[31626]: 2020-04-06T09:47:57.160-07:00 - DEBUG - found 144417 indexed blocks
Apr 06 09:47:57 bolt electrs[31626]: 2020-04-06T09:47:57.490-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:52:28 bolt electrs[31626]: 2020-04-06T09:52:28.712-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:52:52 bolt electrs[31626]: 2020-04-06T09:52:52.003-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:53:14 bolt electrs[31626]: 2020-04-06T09:53:14.931-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:53:39 bolt electrs[31626]: 2020-04-06T09:53:39.975-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:54:02 bolt electrs[31626]: 2020-04-06T09:54:02.422-07:00 - TRACE - downloading 100000 block headers
Apr 06 09:54:27 bolt electrs[31626]: 2020-04-06T09:54:27.782-07:00 - TRACE - downloading 24696 block headers
Apr 06 09:54:43 bolt electrs[31626]: 2020-04-06T09:54:43.660-07:00 - DEBUG - applying 624696 new headers from height 0
Apr 06 09:56:11 bolt electrs[31626]: 2020-04-06T09:56:11.441-07:00 - TRACE - indexed "/mnt/hdd/bc/bitcoin/blocks/blk00000.dat": 0 rows
Apr 06 09:56:22 bolt electrs[31626]: 2020-04-06T09:56:22.190-07:00 - TRACE - indexed "/mnt/hdd/bc/bitcoin/blocks/blk00001.dat": 0 rows
Apr 06 10:00:22 bolt electrs[31626]: 2020-04-06T10:00:22.203-07:00 - TRACE - indexed "/mnt/hdd/bc/bitcoin/blocks/blk00002.dat": 0 rows
Apr 06 10:04:04 bolt electrs[31626]: 2020-04-06T10:04:03.968-07:00 - TRACE - indexed "/mnt/hdd/bc/bitcoin/blocks/blk00003.dat": 1717005 rows
Apr 06 10:05:49 bolt systemd[1]: electrs.service: Main process exited, code=killed, status=9/KILL
Apr 06 10:05:49 bolt systemd[1]: electrs.service: Unit entered failed state.
Apr 06 10:05:49 bolt systemd[1]: electrs.service: Failed with result 'signal'.
Apr 06 10:06:19 bolt systemd[1]: electrs.service: Service hold-off time over, scheduling restart.
Apr 06 10:06:20 bolt systemd[1]: Stopped Electrs daemon.

Stadicus, 2020-04-07:
Not sure why the process is killed after 20 minutes.

* What hardware are you using? (Raspberry Pi 3 or 4, or something else?
* How much memory does it have?

You can always stop the systemd service with `sudo systemctl stop electrs` and then start Electrs manually according to the guide.

3ntranced, 2020-04-07:
Its a Pi 3 with 1 GB. I ll try to see what happens if I run it manually.

3ntranced, 2020-04-07:
Could I be running out of space?
'''/dev/root         14883M   3978M    10269M  28% /
devtmpfs            484M      0M      484M   0% /dev
tmpfs               488M      0M      488M   0% /dev/shm
tmpfs               488M     44M      445M   9% /run
tmpfs                 5M      1M        5M   1% /run/lock
tmpfs               488M      0M      488M   0% /sys/fs/cgroup
/dev/sda1        469325M 423607M    45602M  91% /mnt/hdd
/dev/mmcblk0p1       43M     23M       21M  53% /boot
tmpfs                98M      0M       98M   0% /run/user/1001
'''

3ntranced, 2020-04-07:
this is the error
```
memory allocation of 134210361 bytes failedAborted
```
anything I can do? I think I ve stopped other services, only bitcoind is running

Stadicus, 2020-04-07:
Electrs is very likely crashing because the Raspberry Pi 3 does not have enough memory. My guide is written for the RPi4 with 4GB memory. 

Luckily, there is an option for low-memory devices (but that assumes 2GB memory). I'm not sure Electrs will work with a Raspberry Pi 3... 

See Electrs repository: https://github.com/romanz/electrs/blob/master/doc/usage.md#usage

> If initial sync fails due to memory allocation of xxxxxxxx bytes failedAborted errors, as may happen on devices with limited RAM, try the following arguments when starting electrs. It should take roughly 18 hours to sync and compact the index on an ODROID-HC1 with 8 CPU cores @ 2GHz, 2GB RAM, and an SSD using the following command:
> 
> ```
> $ cargo run --release -- -vvvv --index-batch-size=10 --jsonrpc-import --db-dir ./db --electrum-rpc-addr="127.0.0.1:50001"
> ```

Stadicus, 2020-04-07:
The Electrum Personal Server bonus guide is likely the better solution for a device with just 1GB memory:
https://stadicus.github.io/RaspiBolt/raspibolt_64_electrum.html

3ntranced, 2020-04-09:
> Electrs is very likely crashing because the Raspberry Pi 3 does not have enough memory. My guide is written for the RPi4 with 4GB memory.
>
> Luckily, there is an option for low-memory devices (but [...]

I couldnt figure out how to get this to work. I did edit electrs.conf from this guide to reduce index-batch-size to 10 but it still didnt work. I ll look into upgrading hardware. Thanks

closed 2020-04-09 by 3ntranced