raspibolt/raspibolt #590: error: Could not connect to the server 127.0.0.1:8332 (error code 0 - "timeout reached")

raindogdance, 2020-04-08:
I am getting the following error message when trying to run the command `bitcoin-cli gettxoutsetinfo`:  

`error: Could not connect to the server 127.0.0.1:8332 (error code 0 - "timeout reached") 

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.`

Googling the error message, I found this piece of info: https://github.com/bitcoin/bitcoin/issues/17117#issuecomment-579736563. 

So perhaps "applying rpcclienttimeout=0 to the bitcoin-cli binary" might help here, but how would I go about that? I tried it with `bitcoin-cli -rpcclienttimeout=0 gettxoutsetinfo` but that didn't help. 

Or any other ideas I might try to fix it?  

Btw here is my current bitcoin.conf file, in case that's important. I tried adding `rpcclienttimeout=0` there as well, even if it doesn't seem like the right thing to do:  

# Example bitcoin.conf for RaspiBolt
server=1
daemon=1

txindex=1
disablewallet=0

# use "old" Segregated Witness rpc method until Eclair supports
# the new native Segwit commands of Bitcoin 0.16
deprecatedrpc=addwitnessaddress

# Connection settings
rpcallowip=192.168.0.0/24
rpcallowip=127.0.0.1
rpcuser=X
rpcpassword=X
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
rpcport=8332
rpcbind=192.168.0.9
rpcbind=127.0.0.1:8332
#rpcbind=127.0.0.1
rpcclienttimeout=0

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=150
maxconnections=40
maxuploadtarget=5000

# TOR settings
proxy=127.0.0.1:9050
bind=127.0.0.1
listenonion=1

raindogdance, 2020-04-08:
Updating Bitcoin Core from 0.18 to 0.19.1 seems to have helped! Getting the right output now from the `gettxsetoutinfo` command.

Closing the issue.

closed 2020-04-08 by raindogdance

Stadicus, 2020-04-09:
Thanks for reporting, solving and closing. :-)