raspibolt/raspibolt #591: changing bitcoind tor listening port

3ntranced, 2020-04-11:
I ve set bintcoin.conf parameter -port=10999 but when bitcoind starts I see my onion address followed by 8333. Do I need to make more changes somewhere else?

Stadicus, 2020-04-16:
Can you post the output of the following two commands (without sensitive data like passwords, ip or onion addresses)?

```sh
$ cat /mnt/ext/bitcoin/bitcoin.conf
$ bitcoin-cli getnetworkinfo
```

3ntranced, 2020-04-17:
'''
bitcoin@veg:/home/veg$ bitcoin-cli getnetworkinfo
{
  "version": 190100,
  "subversion": "/Satoshi:0.19.1/",
  "protocolversion": 70015,
  "localservices": "0000000000000409",
  "localservicesnames": [
    "NETWORK",
    "WITNESS",
    "NETWORK_LIMITED"
  ],
  "localrelay": true,
  "timeoffset": -46,
  "networkactive": true,
  "connections": 10,
  "networks": [
    {
      "name": "ipv4",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    },
    {
      "name": "ipv6",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    },
    {
      "name": "onion",
      "limited": false,
      "reachable": true,
      "proxy": "127.0.0.1:9050",
      "proxy_randomize_credentials": true
    }
  ],
  "relayfee": 0.00001000,
  "incrementalfee": 0.00001000,
  "localaddresses": [
    {
      "address": "myveggie.onion",
      "port": 8333,
      "score": 4
    }
  ],
  "warnings": ""
}

3ntranced, 2020-04-17:
```
bitcoin@veg:/home/veg$ cat /mnt/ext/bitcoin/bitcoin.conf
# RaspiBolt: bitcoind configuration
# /mnt/ext/bitcoin/bitcoin.conf

# Bitcoin daemon
server=1
txindex=1

# Network
listen=1
listenonion=1
proxy=127.0.0.1:9050
bind=127.0.0.1:10111

# change port
port=10111

# Connections
rpcuser=raspibolt
rpcpassword=veggiesyum
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
maxconnections=40
maxuploadtarget=5000

# Initial block download optimizations
dbcache=2500
'''
Thanks!

Stadicus, 2020-04-18:
Never tried to change the port for tor onion addess. Is there any reason you'd like to do that? Privacy-wise I don't think it matters. Security-wise it definitely does not.

But you're right, changing the port does not seem to affect the Tor connection. I guess you'd need to set up the Tor onion manually. Locally, I'd keep the standard port, but would then route the incoming traffic from your external Tor port `10111` to the internal ipv4 port `8333`.

relevant configuration in `bitcoin.conf`:
```
listen=1
#listenonion=1     # no longer necessary
proxy=127.0.0.1:9050
bind=127.0.0.1
onlynet=ipv4
```

relevant configuration in `/etc/tor/torrc` (see similar setup for [Electrs](https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html#remote-access)):
```
HiddenServiceDir /var/lib/tor/hidden_service_bitcoind/
HiddenServiceVersion 3
HiddenServicePort 10111 127.0.0.1:8333
```

Your Bitcoin Core client will no longer be aware of its own address and cannot announce it automatically. You could manually announce a regular ip address with `externalip`, not sure if that works with an onion (probably not).

3ntranced, 2020-04-19:
The reason is because I have two rasp pi's behind a router, so the router needs to forward the tor port 8333 of one of them and 10111 of the other one.

Stadicus, 2020-04-19:
That should not be necessary. The beauty of Tor is that it opens a persistent connection from within your network to the Tor network outside. No need to configure any router port forwarding.

This is the same as multiple computer using port 80: if you accept new traffic from the outside, the router needs to know to which computer to forward it to. But if you are opening a new connection from your computer to the outside, returning traffic is automatically routed to the right computer.

You can have as many Tor .onion services on the same port 8333 within your network as you wish. There shouldn't be any conflicts.

3ntranced, 2020-04-20:
nice thank you

closed 2020-04-20 by 3ntranced