raspibolt/raspibolt #593: Possible way to recovery lnd caused by power failure? (config files lost)

Lavande, 2020-04-23:
Hi, my node was recently powered off unexpectedly due to power failure. When I restart raspberry pi, there were I/O errors with the external hard disk. So I did a fsck to recover the filesystem. All is OK but the lnd directory was lost. I only found the following files which are saved.

> bitcoin@LN-Pi:~ $ ls -la /mnt/hdd/lost+found/#21372931/chain/bitcoin/mainnet/
total 28
drwx------ 2 bitcoin bitcoin 4096 Dec 22 17:24 .
drwx------ 3 bitcoin bitcoin 4096 Feb 23  2019 ..
-rw-r--r-- 1 bitcoin bitcoin   83 Dec 21 16:17 chainnotifier.macaroon
-rw-r--r-- 1 bitcoin bitcoin 2348 Dec 22 17:24 channel.backup
-rw-r--r-- 1 bitcoin bitcoin   91 Dec 21 16:17 invoices.macaroon
-rw-r--r-- 1 bitcoin bitcoin   86 Dec 21 16:17 signer.macaroon
-rw-r--r-- 1 bitcoin bitcoin  114 Dec 21 16:17 walletkit.macaroon

Is it possible to recovery my lnd node from only these files?

Lavande, 2020-04-26:
Solved with this doc:
https://github.com/lightningnetwork/lnd/blob/master/docs/recovery.md

closed 2020-04-26 by Lavande

Stadicus, 2020-04-27:
Oh awesome, thanks for sharing!