raspibolt/raspibolt #595: Why do I have to update and why am I unreachable?

miguelmartinn9, 2020-04-24:
So a few months ago I unplugged my bitcoin node when it was running perfectly. So when I connected it again this week I had to let it sync to the blockchain again. Now you can see it's synced but I have several issues. 
It tells me to update the version when v0.19.0.1 is totally up to date. And if I need to update, how do I do it? 
Secondly I don't know but it says that my node is unreachable (I cropped out the IP address). How can I fix it? 

Thanks
[image]

Xekyo, 2020-04-25:
The latest release of Bitcoin Core is [`v0.19.1`](https://bitcoin.org/en/release/v0.19.1) which is newer than the `v0.19.0.1` you're running. The [RaspiBolt FAQ](https://stadicus.github.io/RaspiBolt/raspibolt_faq.html#how-to-upgrade-bitcoin-core) gives instructions on how to update.

I don't know about the "Not reachable".

miguelmartinn9, 2020-04-25:
Thanks about the update. The unreachable thing it's because I am having problems with my ISP and port forwarding. Thanks!

53645714n, 2020-04-25:
Did you try the bitnodes way? I added it to the motd file, just have to fiddle with comment tags, it's in the tutorial.

miguelmartinn9, 2020-04-25:
Nope, I haven't what is there to try?

53645714n, 2020-04-25:
There's multiple ways of checking if your node is reachable, one is via pinging your own address which is failing, the second is via bitnodes.earn. Both ways are programmed in the motd, one of them is commented out.
If you have incoming connections however, you don't need to worry. That is the best indicator.

miguelmartinn9, 2020-04-26:
I have one last question. If I don't have any incoming connections (which I don't, I only have outgoing) can I connect my node to Electrum like it says [here](https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html#connect-electrum-wallet)?

closed 2020-04-28 by miguelmartinn9