raspibolt/raspibolt #598: txindex is enabled at height 625874 what does it mean?

3ntranced, 2020-05-01:
Why doesnt it say its enabled at the latest height? What does it mean please?

Stadicus, 2020-05-04:
Can you please provide some context? I have no idea what section or configuration you are referring to.

3ntranced, 2020-05-05:
sorry. looking at debug.log for bitcoind. trying to figure out if "txindex is up to date", if there is such a thing.

```
2020-05-05T06:08:38Z Using obfuscation key for /mnt/ext/bitcoin/indexes/txindex: 0000000000000000
2020-05-05T06:08:38Z txindex thread start
2020-05-05T06:08:38Z txindex is enabled at height 628983
2020-05-05T06:08:38Z txindex thread exit
2020-05-05T06:10:21Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
```

closed 2020-05-25 by 3ntranced