raspibolt/raspibolt #599: electrs fails when making full compaction

dk88888, 2020-05-01:
2020-05-01T19:28:52.241+02:00 - DEBUG - no more blocks to index
2020-05-01T19:28:56.795+02:00 - DEBUG - no more blocks to index                                                          
2020-05-01T19:28:56.796+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02054.dat": 1556904 rows
2020-05-01T19:29:01.940+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02056.dat": 99138 rows
2020-05-01T19:29:01.940+02:00 - DEBUG - no more blocks to index
2020-05-01T19:29:02.365+02:00 - DEBUG - no more blocks to index
2020-05-01T19:29:02.366+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02055.dat": 1538478 rows
2020-05-01T19:29:08.044+02:00 - DEBUG - last indexed block: best=000000000000000000061cd5afc6b6bce364cbc63b92b3ac2f9a3cd57037bf64 height=628430 @ 2020-05-01T15:41:03Z                                                                            
2020-05-01T19:29:10.482+02:00 - INFO - starting full compaction
2020-05-01T21:33:55.205+02:00 - INFO - finished full compaction                                                          
2020-05-01T21:33:55.214+02:00 - INFO - enabling auto-compactions                                                         
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { message: "Result incomplete: IO error: No space left on deviceWhile appending to file: /mnt/ext/electrs/db/mainnet/000607.sst: No space left on device" }', src/store.rs:159:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
2020-05-01T21:33:55.276+02:00 - TRACE - closing DB at "/mnt/ext/electrs/db/mainnet"

dk88888, 2020-05-01:
Before starting of reindexing I have ~112 gb of free space and after hours of process I got this error.

3ntranced, 2020-05-01:
I too was running out of space. 500GB is not enough for block chain and electrs. I had to get 1TB. Maybe if you 550 GB that should be enough for current sized blockchain and for electrs to do its compaction.

Stadicus, 2020-05-04:
Yes, electrs needs a over 100 GB before compacting the database down to about 50GB. It's probably time to update recommentation to use a 1TB drive.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus