raspibolt/raspibolt #600: what does it mean when two of my peers are from 127.0.0.1?

3ntranced, 2020-05-02:
``

11 | /Satoshi:0.19.1/ | 127.0.0.1:57856 | 0000000000000409 | 0:02 / 0:02
-- | -- | -- | -- | --
/Satoshi:0.19.0.1/ | 127.0.0.1:39460 | 0000000000000409 | 0:02 / 0:03

``

3ntranced, 2020-05-03:
and in debug.log I am getting the following entries
2020-05-03T05:19:16Z ThreadRPCServer incorrect password attempt from 127.0.0.1:37494
2020-05-03T05:20:20Z ThreadRPCServer incorrect password attempt from 127.0.0.1:37558

Stadicus, 2020-05-04:
Are you running the BTC RPC Explorer, or Electrs or additional software?

3ntranced, 2020-05-04:
Yes BTC RPC explorer and Electrs. Is it them?

Stadicus, 2020-05-05:
Probably. You should be able to see unsuccessful connection attempts in the logfiles of either of these two services. Or you just stop one, and see if the errors in the bitcoind log stop.

closed 2020-05-05 by 3ntranced