raspibolt/raspibolt #602: Could not connect to Electrum Personal Servis

dk88888, 2020-05-03:
My laptop's Electrum client could not connect to my EPS. Both of these devices are on the same local network. Also `eps.service` is on and rescan completed. 
Trying to connect my Electrum with verbose logging enabled:
`electrum -v --oneserver --server 192.168.1.129:50002:s` and get that output:
```
I | logging | Logging to file: None
I | logging | Log filters: verbosity '*', verbosity_shortcuts ''
I | daemon | failed to connect to JSON-RPC server: [Errno 111] Connection refused
...
D | util.profiler | Plugins.__init__ 0.0055
I/n | network | blockchains [0]
I | exchange_rate.FxThread | using exchange CoinCap
I/n | network | starting network
I/n | network | connecting to 192.168.1.129:50002:s as new interface
I | exchange_rate.CoinCap | getting fx quotes for USD
D | util.profiler | Daemon.__init__ 0.0039
I/i | interface.[192.168.1.129:50002] | disconnecting due to: CancelledError()
I/n | network | couldn't launch iface 192.168.1.129:50002:s -- CancelledError()
I | exchange_rate.CoinCap | received fx quotes
D | util.profiler | ElectrumGui.__init__ 0.1605
I | storage.WalletStorage | wallet path /home/dk/.electrum/wallets/default_wallet2
D | util.profiler | JsonDB._load_transactions 0.0000
D | util.profiler | AddressSynchronizer.load_local_history 0.0000
D | util.profiler | AddressSynchronizer.check_history 0.0002
D | util.profiler | AddressList.update 0.0057
D | util.profiler | Deterministic_Wallet.try_detecting_internal_addresses_corruption 0.0265
D | util.profiler | ElectrumWindow.load_wallet 0.0587
I | gui.qt.history_list.HistoryModel | refreshing... reason: update_tabs
D | util.profiler | Abstract_Wallet.get_full_history 0.0003
D | util.profiler | HistoryModel.refresh 0.0011
D | util.profiler | AddressList.update 0.0173
I/n | network | connecting to 192.168.1.129:50002:s as new interface
I/i | interface.[192.168.1.129:50002] | disconnecting due to: CancelledError()`
```
I tried to narrow my problem running `ncat` with `-v` for verbose output: 

- Query EPS through nginx provides that output:

```
$ ncat -v --ssl localhost 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.
Ncat: Trying next address...
Ncat: Subject: CN=localhost
Ncat: Issuer: CN=localhost
Ncat: SHA-1 fingerprint: C0A7 8C7A 472E 46AD 8A3F B8E0 F824 0736 8A6F 5E92
Ncat: Certificate verification failed (self signed certificate).
Ncat: SSL connection to 127.0.0.1:50002.
Ncat: SHA-1 fingerprint: C0A7 8C7A 472E 46AD 8A3F B8E0 F824 0736 8A6F 5E92
```
How can I fix this connection issue?

Stadicus, 2020-05-04:
These instructions work for Electrs, buc can you try if they work on EPS as well?
https://github.com/Stadicus/RaspiBolt/issues/565#issuecomment-598953295

dk88888, 2020-05-04:
> These instructions work for Electrs, buc can you try if they work on EPS as well?
> [#565 (comment)](https://github.com/Stadicus/RaspiBolt/issues/565#issuecomment-598953295)

I tried these instructions but they doesn't work for me. Still have this connection issue.

Stadicus, 2021-11-27:
Unfortunately, EPS is no longer supported with the latest guide update to version 3. This is because Bitcoin Core needs to run with the wallet disabled to allow for proper access privileges.

I recommend setting up Electrs, which is much more convenient once it's synced.

Closing this issue due for now. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus