raspibolt/raspibolt #604: After the TOR section : not finding my Raspberry on the network

blockchaineur, 2020-05-16:
Hi, after installing TOR and rebooting, I can't find and connect with putty to the raspberry pi (maybe because of TOR routing to a new IP adress?). I tried multiples things but I can't connect to it, could you help?

Stadicus, 2020-05-25:
Tor should not interfere with regular `ssh` access, as it's opening a connection to the Tor network from the Pi to the outside, not blocking anything coming in. Maybe it's the `ufw` config that blocks port `22`?

If you can't connect, hook up the Pi to a screen and keyboard, then you can login locally without any network dependencies.

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus