raspibolt/raspibolt #607: Permission error while trying to configure Lightning Node

miguelmartinn9, 2020-05-28:
A few months ago I finished implementing the Bitcoin Node and now I wanted to add the LN one. 

I installed it without problems but at the very start of the Configuration Section, I try to create the `lnd` folder inside `/mnt/ext`. With the following command (and as bitcoin user):
`mkdir /mnt/ext/lnd`.

And then the error says:
`mkdir: cannot create directory ‘/mnt/ext/lnd’: Permission denied`

If I try to do it with `sudo`, then it says that bitcoin is not in the sudoers file. 

Can anyone help me?

Thanks

miguelmartinn9, 2020-05-31:
Same thing happens when trying to configure Electrs

53645714n, 2020-05-31:
If it is just the creaton of the folders, do it as the pi or admin user and go from there.

closed 2020-05-31 by miguelmartinn9