raspibolt/raspibolt #609: WARN - reconnecting to bitcoind: no reply from daemon

miguelmartinn9, 2020-05-31:
When I try to run Electrs manually it gives me this warning constantly:
` WARN - reconnecting to bitcoind: no reply from daemon`
I don't know what it could be since when I run: `service bitcoind status`
It tells me it's totally active

miguelmartinn9, 2020-05-31:
This is what it appears when I first run the command, before the WARN 
Config { 
log: StdErrLog { verbosity: Trace, quiet: false, timestamp: Millisecond, modules: [], writer: "stderr", color_choice: Auto }, network_type: bitcoin, db_path: "/mnt/ext/electrs/db/mainnet", daemon_dir: "/mnt/ext/bitcoin", daemon_rpc_addr: V4(127.0.0.1:8332), electrum_rpc_addr: V4(127.0.0.1:50001), monitoring_addr: V4(127.0.0.1:4224), jsonrpc_import: false, index_batch_size: 100, bulk_index_threads: 4, tx_cache_size: 10485760, txid_limit: 1000, server_banner: "Welcome to electrs 0.8.3 (Electrum Rust Server)!", blocktxids_cache_size: 10485760 }

miguelmartinn9, 2020-06-01:
I am a dumb fuck and I forgot to put "raspibolt:..." in the config file.

Stadicus, 2020-06-03:
Let me know if you need additional help

miguelmartinn9, 2020-06-04:
Alright, thanks a lot

closed 2020-06-04 by miguelmartinn9