raspibolt/raspibolt #610: memory allocation of 134117104 bytes failedAborted

miguelmartinn9, 2020-06-01:
When I try to manually run Electrs to check that everything is alright (un)surprisingly everything is not alright. 
I run the command 
`electrs --conf /mnt/ext/electrs/electrs.conf`

And then it goes like this:
`2020-06-01T22:07:16.329+02:00 - DEBUG - Server listening on 127.0.0.1:4224
2020-06-01T22:07:16.333+02:00 - DEBUG - Running accept thread
2020-06-01T22:07:16.334+02:00 - INFO - NetworkInfo { version: 190100, subversion: "/Satoshi:0.19.1/" }
2020-06-01T22:07:16.337+02:00 - INFO - BlockchainInfo { chain: "main", blocks: 632643, headers: 632643, bestblockhash: "0000000000000000000398d4a52962ab47f8b8ac4d36eec5ed7bcef33e8b1b3a", pruned: false, initialblockdownload: false }
2020-06-01T22:07:16.340+02:00 - DEBUG - opening DB at "/mnt/ext/electrs/db/mainnet"
2020-06-01T22:07:16.540+02:00 - TRACE - lastest indexed blockhash: 0000000000000000000000000000000000000000000000000000000000000000
2020-06-01T22:07:16.541+02:00 - INFO - listing block files at "/mnt/ext/bitcoin/blocks/blk*.dat"
2020-06-01T22:07:16.668+02:00 - INFO - indexing 2097 blk*.dat files
2020-06-01T22:07:16.668+02:00 - DEBUG - found 0 indexed blocks
2020-06-01T22:07:16.681+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:07:38.625+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:07:58.675+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:08:19.158+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:08:39.605+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:08:59.644+02:00 - TRACE - downloading 100000 block headers
2020-06-01T22:09:20.350+02:00 - TRACE - downloading 32644 block headers
2020-06-01T22:09:38.184+02:00 - DEBUG - applying 632644 new headers from height 0
2020-06-01T22:17:32.773+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00003.dat": 1717278 rows
2020-06-01T22:19:30.276+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00002.dat": 1742521 rows
2020-06-01T22:32:20.156+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00001.dat": 1651705 rows
2020-06-01T22:36:07.600+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00000.dat": 1712063 rows
2020-06-01T22:43:14.794+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00004.dat": 1688621 rows
2020-06-01T22:50:45.923+02:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00005.dat": 1621538 rows
`
And finally the error goes off and says:
`memory allocation of 134117104 bytes failedAborted`

Why does this happen? 
Thanks!

miguelmartinn9, 2020-06-01:
I think it may be because I am running this on a raspberry pi 3 and I don't have a lot of RAM. But I don't really know

Stadicus, 2020-06-03:
Yes, a RPi3 does not have enough memory to run Electrs. Try using Electrum Personal Server (see [bonus guide](https://stadicus.github.io/RaspiBolt/raspibolt_64_electrum.html) or the new [Bitcoin Wallet Tracker](https://twitter.com/Stadicus3000/status/1266434524822544387) (needs compilation).

miguelmartinn9, 2020-06-04:
Will do, thanks

miguelmartinn9, 2020-06-04:
Hey @Stadicus I succesfully set up Electrum Personal Server (yay!) but know bitnodes says that my node is unreachable. Why is that? 
[image]

miguelmartinn9, 2020-06-04:
I only have outgoing connections, and nobody is connected to me. How is this possible?

miguelmartinn9, 2020-06-04:
It looks like it's a problem of UFW. When I disable it, then it tells me it's reachable.

Stadicus, 2020-06-05:
What does `sudo ufw status` say?

miguelmartinn9, 2020-06-05:
This:
(It says active, because I enabled it right now, but my node is only reachable  when its disabled)
`Status: active

To                         Action      From
--                         ------      ----
22                         ALLOW       Anywhere                   # allow SSH
50002                      ALLOW       Anywhere                   # allow Electrum SSL
50002                      ALLOW       192.168.0.0/24             # allow EPS from local network
22 (v6)                    ALLOW       Anywhere (v6)              # allow SSH
50002 (v6)                 ALLOW       Anywhere (v6)              # allow Electrum SSL
`

Stadicus, 2020-06-05:
Ah, you need to open port `8333` if you want to allow incoming clearnet connections. This is the only difference UFW makes.

The current RaspiBolt guide is written for running on Tor only. Are you deliberately running on clearnet and not Tor? 

If you want to use Tor only, then no need to open any additional ports. Otherwise, allow the port with `sudo ufw allow 8333  comment 'allow Bitcoin mainnet'`.

miguelmartinn9, 2020-06-05:
I thought I was running on Tor. Is there any way to check it?

miguelmartinn9, 2020-06-05:
I mean, if I run on TOR then it's normal that I have no incoming connections?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus