raspibolt/raspibolt #611: Rebooted my Raspibolt after the Privacy step, now unable to login over SSH or locally

SnakeOfMidgaard, 2020-06-03:
Hello!
I've been following the guide for setting this up the last few days. I had just finished the Privacy part of the guide and figured it was time for a simple reboot of my Pi before continuing with the Bitcoin part.

Ever since that restart whenever i try to SSH into the Raspibolt it asks for username and password and immediately closes the Putty session when entering the correct password.
I now connected it to a display and keyboard, and checked it.
It boots up and asks for username/pw but gives me the error "cannot make/remove an entry for the specified session" when entering the correct password.

I've looked through the files i have edited throughout the guide on a separate Linux install and everything seems to be correct. The only thing i've noticed is that my "/etc/security/etc/pam.d/ files aren't identical in terms of comments in the two files compared to the video in the guide.
Maybe that part needs an update?
Any other ideas?

Thanks!

Stadicus, 2020-06-03:
Can you locally log in as user "root"?

SnakeOfMidgaard, 2020-06-03:
> 
> 
> Can you locally log in as user "root"?

Unfortunately not. It gives me the same error of "Cannot make/remove an entry for the specified session".

> 
> 
> On a new install, everything in `/etc/security/limits.conf` is commented out. If you can access the file system, maybe try resetting this?

I resetted the file (removing the soft/hard and root soft/root hard lines entirely) and tried again; same error as above.

> 
> If necessary, you can try resetting the other two files `/etc/pam.d/common-session` and `/etc/pam.d/common-session-noninteractive` as well. Let me know if you need the defaults or more info.

I resetted both files to their default and *finally*! This did it!
I can now both SSH into it and log in locally as admin(and the other users).

Thank you for the very quick help!
Question is; can i finish setting this up or should we figure out how to make it work with these files?

SnakeOfMidgaard, 2020-06-03:
Here are what the two files look like by default in newest image of Raspbian, which i'm on:

common-session

```
# /etc/pam.d/common-session - session-related modules common to all services
#
# This file is included from other service-specific PAM config files,
# and should contain a list of modules that define tasks to be performed
# at the start and end of sessions of *any* kind (both interactive and
# non-interactive).
#
# As of pam 1.0.1-6, this file is managed by pam-auth-update by default.
# To take advantage of this, it is recommended that you configure any
# local modules either before or after the default block, and use
# pam-auth-update to manage selection of other modules.  See
# pam-auth-update(8) for details.

# here are the per-package modules (the "Primary" block)
session [default=1]                     pam_permit.so

# here's the fallback if no module succeeds
session requisite                       pam_deny.so
# prime the stack with a positive return value if there isn't one already;
# this avoids us returning an error just because nothing sets a success code
# since the modules above will each just jump around
session required                        pam_permit.so
# and here are more per-package modules (the "Additional" block)
session required        pam_unix.so
session optional        pam_systemd.so
session optional        pam_chksshpwd.so

# end of pam-auth-update config
```

common-session-noninteractive

```
# /etc/pam.d/common-session-noninteractive - session-related modules
# common to all non-interactive services
#
# This file is included from other service-specific PAM config files,
# and should contain a list of modules that define tasks to be performed
# at the start and end of all non-interactive sessions.
#
# As of pam 1.0.1-6, this file is managed by pam-auth-update by default.
# To take advantage of this, it is recommended that you configure any
# local modules either before or after the default block, and use
# pam-auth-update to manage selection of other modules.  See
# pam-auth-update(8) for details.

# here are the per-package modules (the "Primary" block)
session [default=1]                     pam_permit.so
# here's the fallback if no module succeeds
session requisite                       pam_deny.so
# prime the stack with a positive return value if there isn't one already;
# this avoids us returning an error just because nothing sets a success code
# since the modules above will each just jump around
session required                        pam_permit.so
# and here are more per-package modules (the "Additional" block)
session required        pam_unix.so

# end of pam-auth-update config
```

Stadicus, 2020-06-04:
> Question is; can i finish setting this up or should we figure out how to make it work with these files?

You can just go ahead, I will look deeper into this issue. It's an optional hardening measure, and the Pi will run just fine without it.

SnakeOfMidgaard, 2020-06-04:
> 
> 
> > Question is; can i finish setting this up or should we figure out how to make it work with these files?
> 
> You can just go ahead, I will look deeper into this issue. It's an optional hardening measure, and the Pi will run just fine without it.

Okay! Thank you, again.
Let's get this node up and running.

closed 2021-11-27 by Stadicus