raspibolt/raspibolt #622: can't upgrade to lnd 0.10.1 - keybase.io cert. expired / not trusted

gitro83, 2020-06-21:
Hi, 

I can't upgrade from lnd 0.10.0 to 0.10.1 / can't verify pgp, this s what happens (relevant part):

`wget https://keybase.io/roasbeef/pgp_keys.asc
--2020-06-21 09:43:06--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 50.17.252.94, 35.169.49.45
Connecting to keybase.io (keybase.io)|50.17.252.94|:443... connected.
ERROR: The certificate of 'keybase.io' is not trusted.
ERROR: The certificate of 'keybase.io' has expired.`

`gpg ./pgp_keys.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: can't open './pgp_keys.asc'`

Thanks in advance for any help.

tvial-tagheuer, 2020-06-21:
Having same issue (fr locale)

```
bitcoin@lnd01:~/download $ wget https://keybase.io/roasbeef/pgp_keys.asc
--2020-06-21 21:06:41--  https://keybase.io/roasbeef/pgp_keys.asc
Résolution de keybase.io (keybase.io)… 50.17.252.94, 35.169.49.45
Connexion à keybase.io (keybase.io)|50.17.252.94|:443… connecté.
Erreur : le certificat de « keybase.io » n’est pas de confiance.
Erreur : le certificat de « keybase.io » a expiré.
```

lucasmoten, 2020-06-21:
I ran into this as well for
```
wget https://keybase.io/roasbeef/pgp_keys.asc
```

As a temporary workaround I can pull with curl, 
```
curl -v -O https://keybase.io/roasbeef/pgp_keys.asc
```
The verbose dump may show some clues

```
 Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 0 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 1 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 4 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 2 ms for 1 (transfer 0x1646880)
* Expire in 3 ms for 1 (transfer 0x1646880)
*   Trying 50.17.252.94...
* TCP_NODELAY set
* Expire in 149995 ms for 3 (transfer 0x1646880)
* Expire in 200 ms for 4 (transfer 0x1646880)
* Connected to keybase.io (50.17.252.94) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [102 bytes data]
* TLSv1.2 (IN), TLS handshake, Certificate (11):
{ [4621 bytes data]
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
{ [333 bytes data]
* TLSv1.2 (IN), TLS handshake, Server finished (14):
{ [4 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
} [70 bytes data]
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.2 (OUT), TLS handshake, Finished (20):
} [16 bytes data]
* TLSv1.2 (IN), TLS handshake, Finished (20):
{ [16 bytes data]
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: OU=Domain Control Validated; OU=PositiveSSL; CN=keybase.io
*  start date: May 20 00:00:00 2019 GMT
*  expire date: Jun 16 23:59:59 2021 GMT
*  subjectAltName: host "keybase.io" matched cert's "keybase.io"
*  issuer: C=GB; ST=Greater Manchester; L=Salford; O=Sectigo Limited; CN=Sectigo RSA Domain Validation Secure Server CA
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
* Using Stream ID: 1 (easy handle 0x1646880)
} [5 bytes data]
> GET /roasbeef/pgp_keys.asc HTTP/2
> Host: keybase.io
> User-Agent: curl/7.64.0
> Accept: */*
> 
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
} [5 bytes data]
< HTTP/2 200 
< date: Sun, 21 Jun 2020 19:57:14 GMT
< content-type: text/plain; charset=utf-8
< content-length: 6909
< x-frame-options: SAMEORIGIN
< x-xss-protection: 1; mode=block
< x-content-type-options: nosniff
< access-control-allow-origin: *
< access-control-allow-methods: GET
< access-control-allow-headers: Content-Type, Authorization, Content-Length, X-Requested-With
< access-control-allow-credentials: false
< cache-control: no-store

```

lucasmoten, 2020-07-11:
As a follow up, I tried the steps this evening with wget when upgrading to LND v0.10.3 and it worked fine
```
$ wget https://keybase.io/roasbeef/pgp_keys.asc
--2020-07-11 03:23:39--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 52.20.35.169, 52.204.151.216
Connecting to keybase.io (keybase.io)|52.20.35.169|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6909 (6.7K) [text/plain]
Saving to: ‘pgp_keys.asc’

pgp_keys.asc                             100%[===============================================================================>]   6.75K  --.-KB/s    in 0.005s  

2020-07-11 03:23:40 (1.36 MB/s) - ‘pgp_keys.asc’ saved [6909/6909]
```

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus