raspibolt/raspibolt #624: rpc error: code = Unknown desc = server is still in the process of starting

nipponese, 2020-07-03:
The LND log shows the chain in synced:
`2020-07-03 09:38:33.297 [INF] LTND: Chain backend is fully synced (end_height=637528)!`

However, I cannot connect to other nodes:
```$ lncli connect 03d06758583bb5154774a6eb221b1276c9e82d65bbaceca806d90e20c108f4b1c7@gwdllz5g7vky2q4gr45zguvoajzf33czreca3a3exosftx72ekppkuqd.onion
[lncli] rpc error: code = Unknown desc = server is still in the process of starting
```

FYI, this is the yalls node.

Stadicus, 2020-07-08:
Not quite sure what's happening here. Maybe this is more an issue for the LND project?
https://github.com/lightningnetwork/lnd/

nipponese, 2020-07-08:
Hmm, yeah, eventually it went away and now I can use lncli normally.

closed 2021-11-27 by Stadicus