raspibolt/raspibolt #626: creating a backup of blockchain and index

3ntranced, 2020-07-09:
Since it takes such a long time to reindex the blockchain. Whats the best way to make a backup of blockchain, indexes, chainstate, etc? Can I just use rsync while bitcoind is running? Or do I need to stop bitcoind? Is there a way to do it without stopping bitcoind?

3ntranced, 2020-07-10:
I changed my hdd enclosure, bitcoind gave me a corruption error of my chainstates so had to wait 2-3 days for -reindex-chainstate to complete. Having a backup would be more time efficient.

lucasmoten, 2020-07-11:
I'd recommend stopping the service prior to any copying of the directory

3ntranced, 2020-07-13:
I also got this response on bitcoin github
https://github.com/bitcoin/bitcoin/issues/19487#issuecomment-657615767

3ntranced, 2020-07-13:
> @3ntranced It's an important issue but will take a while to implement. In the meantime: I backup the Bitcoin blockchain without stopping bitcoind every month on my server (basically) like this:
> 
> ```
> bitcoin-0.20.0/src/bitcoin-cli savemempool
> rsync -L --delete -av ".bitcoin/" /media/external/.bitcoin
> rsync -L --delete -av ".bitcoin/" /media/external/.bitcoin
> rsync -L --delete -av ".bitcoin/" /media/external/.bitcoin
> rsync -L --delete -av ".bitcoin/" /media/external/.bitcoin
> ```
> 
> Usually rsync doesn't do anything anymore for the last 2 executions. I then run bitcoin-qt on a different computer (but you could run it on your server too) like this (connecting only to my own bitcoind):
> `bitcoin-qt -connect=192.168.178.20 -blocksonly=1 -listen=0 -datadir=/media/external/.bitcoin`
> 
> Bitcoin-qt always repairs the blockchain without problems and this way you'll know if the backup was successful. You can even stop bitcoin-qt cleanly and run it again.

lucasmoten, 2020-07-14:
I'm going to be upgrading the drive on one of the instances I have running over the next week or two.  When I do that, I'll document the steps I take.  My plan is to  
- [ ] stop and disable services
- [ ] unwind/move aside any login scripts that depend on services (e.g. 61 system overview)
- [ ] mount the new drive to /mnt/extnew defined in /etc/fstab alongside old drive
- [ ] rsync files from old drive to new
- [ ] unmount both external drives
- [ ] update /etc/fstab to set new in place of old on /mnt/ext
- [ ] mount the new drive
- [ ] verify file ownership is set
- [ ] recreate swap file
- [ ] start services
- [ ] verify service communication
- [ ] restore system overview

3ntranced, 2020-07-17:
whats the reason for recreating swap file? what does it do?

3ntranced, 2020-08-09:
the above commands work really well for backing up the blockchain.

closed 2020-08-09 by 3ntranced