raspibolt/raspibolt #630: Password for Block explorer + pinned server Electrum app - Proxy tab in Electrum - Updating fix

2pac1, 2020-07-12:
Hey guys,

I would appreciate some eyes on https://github.com/Stadicus/RaspiBolt/pull/621
Everything has been tested by me except password for the Block explorer.

Info:
-added info on how to run Electrum on Mac in the form of an .app
-added info on how to make sure you get direct connection to your node in case you have been previously using TOR under Proxy tab in Electrum
-added section on how to proceed with updating Electrs if you have files that are changed locally

lucasmoten, 2020-07-14:
I was looking that over as I also have a PR in on the electrs for version update, as well as sample output update.

I don't have a mac to verify your approach, but it doesn't break anything and can add some guidance.  

Since this is Stadicus repository, he's the ultimate gatekeeper, but I do have some thoughts..

- Personal preference - I tend to favor single commits (squash and force push to the same branch and then PR), as it makes for cleaner history
- Personal preference - This PR appears to address multiple concerns that may be better broken up into separate PRs
- The following statement in raspibolt_50_electrs.md appears to go against the intent to run as TOR as setup in raspibolt_22_privacy
  > If you have previously been connecting over TOR, make sure everything in Proxy tab is unchecked so direct connection to Raspibolt on your LAN can be made
- The first reference to BONUS INFO, for those using a mac is likely helpful, but it may more appropriate titled as Mac specific.  It would also be more helpful if it gave the sample for connecting through TOR as well in the section that follows
- The second reference to BONUS INFO is worrisome to me.  What is being referred to by this:
  > When installing Electrs, you might notice alert in terminal warning you about insecurities in some of the files that are being downloaded
- The guidance for raspibolt_55_explorer about enabling basic authentication is, as I understand it, simple Base64 encoding of plaintext.

2pac1, 2020-07-14:
Hey,

Thank you for your feedback.
I agree that separate commits are preferable.
I just bundled this up not to forget it and haven't sepersted.

On the proxy tab, it might need more direct explanation that if you were
using TOR previously before this guide in electrum, that now if you decide
to use direct IP on local LAN instead of .onion address, that proxy tab
needs to be looked at.
Does that make sense?

For Bonus info, yes, Mac specific.
Correct.
Maybe to give out both depending if you are connecting via IP on LAN or via
TOR and using .onion address.

Electrs, mom audit and security vulnerabilities maybe needs to be better
explained.
I encountered that where I got the mag, updated some non electrum stuff,
and later on couldn't update electrum as some of the stuff has changed
locally.
Maybe even explaining that can cause more confusion.
Just an issue I have encountered.

Oh, so pass is basically non existent.
Got it.

Thank you very much for your feedback.
If you can try to help break these up, so we make it tidy and streamlined
so for example if you want a Mac shortcut app, you can create one and if
you don't, you can skip that paragraph.

Thanks for your time once again.

lucasmoten, 2020-07-15:
> On the proxy tab, it might need more direct explanation that if you were using TOR previously before this guide in electrum, that now if you decide to use direct IP on local LAN instead of .onion address, that proxy tab needs to be looked at. Does that make sense?

Yes. I would make the assumption that anyone following the guide up to this point has done all prior steps, so access would be with TOR by default.  Though that isn't true for anyone doing piecemeal type upgrades of existing deployments.

> Electrs, mom audit and security vulnerabilities maybe needs to be better explained. I encountered that where I got the mag, updated some non electrum stuff, and later on couldn't update electrum as [...]

Interesting.  With Electrs being done in Rust, I wonder if the security warnings you saw were from older version of Rust?  When I did this recently, I used the newer versions of Rust and Electrs (thats what my PR is) to verify they work.  I don't recall seeing any  vulnerability reports or warnings, but then again, the only thing I have running on this unit is the information from this guide.  If you have log output that may be helpful to discern whether issues are benign, or warrant attention

2pac1, 2020-07-15:
All the logs that I have saved internally are a note that when I went to update Bitcoin Block Explorer to the latest version, that then I got this msg:

btcrpcexplorer@raspibolt:~ $ cd ~/btc-rpc-explorer
btcrpcexplorer@raspibolt:~/btc-rpc-explorer $ git fetch
btcrpcexplorer@raspibolt:~/btc-rpc-explorer $ git checkout v2.0.2
error: Your local changes to the following files would be overwritten by checkout:
	npm-shrinkwrap.json
	package.json
Please commit your changes or stash them before you switch branches.
Aborting

At that point I got stuck, as I didn't know why I cant checkout latest version.

When I look at my bash history, I fund that under btcrpcexplorer username I see:
npm install meow@7.0.1 
npm audit - this on sh
npm audit fix

Try this:
1. sudo su - btcrpcexplorer
2. cd ~/btc-rpc-explorer
3. npm audit
>> here you should see something like:

 === npm audit security report ===                        
                                                                                
found 0 vulnerabilities
 in 407 scanned packages

Here at these steps for example is where I saw that there are some vulnerabilities.
That made me do:
npm audit fix
That fix some, but not all.

I did: npm audit 
again and looked for which files needs updating manually, for example:
npm install meow@7.0.1

After that, npm audit showed 0 vulnerabilities.
But that changed my files locally or something, so when I went out to upgrade I couldn't as i got the msg: error: Your local changes to the following files would be overwritten by checkout:
	npm-shrinkwrap.json
	package.json

To fix that I had to do: git reset HEAD --hard
and then once I updated, if I wanted to update these files/vulnerabilities manually, I could do that.

PROXY TAB
Regarding proxy tab.
Consider this scenario.
Prior to starting a Raspibolt guide, you use electrum over TOR. SO your proxy tab is enabled with Use proxy and Use Tor.
Now in latest electrum version it's just Use proxy check.
So when you go through the guide, and you get to the point where you are like Yes, I will use Electrum pinning to my RaspiBolt from my LAN, over local IP, no need to use TOR as I do it from home.
Once you setup your local IP there, it will show No connection and it will not work, if you do not remember to go to Proxy tab and uncheck what is checked there. Does that make sense?

2pac1, 2020-07-16:
Info about what I was seeing: https://docs.npmjs.com/auditing-package-dependencies-for-security-vulnerabilities

lucasmoten, 2020-07-17:
I just upgraded my drive to SSD, and am now able to run the Block Explorer.  There are two menu choices for RPC Browser and RPC Terminal.  Both of these result in the following for me
```
RPC Terminal / Browser require authentication. Set an authentication password via the 'BTCEXP_BASIC_AUTH_PASSWORD' environment variable (see .env-sample file for more info).
```
I'm inclined to leave this disabled as I don't have a use case where I'd want access to those in the browser, in leiu of direct terminal commands on the Pi itself.

How are you using that feature?

lucasmoten, 2020-07-17:
### Startup for Block Explorer 2.0.02

This is what I'm running.  My startup looks like this

```sh
$ npm run start

> btc-rpc-explorer@2.0.2 start /home/btcrpcexplorer/btc-rpc-explorer
> node ./bin/www

Loading env file: /home/btcrpcexplorer/btc-rpc-explorer/.env
  btcexp:app Starting BTC RPC Explorer, v2.0.2 (commit: '0f134133f5', date: 2020-07-03) +0ms
  btcexp:app Connecting to RPC node at 127.0.0.1:8332 +1ms
  btcexp:app Verifying RPC connection... +36ms
  btcexp:app Loading mining pools config +8ms
  btcexp:app RPC Connected: version=200000 subversion=/Satoshi:0.20.0/, parsedVersion(used for RPC versioning)=0.20.0, protocolversion=70015, chain=main, services=[NETWORK, WITNESS, NETWORK_LIMITED] +181ms
  btcexp:app Loading historical data for chain=main +1ms
  btcexp:app Skipping performance-intensive task: fetch UTXO set summary. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details. +1ms
  btcexp:app Skipping performance-intensive task: fetch last 24 hrs of blockstats to calculate transaction volume. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details. +0ms
```

### Switching branches with uncommitted changes

When you are faced with uncommited changes, you can consider either stashing to change branches
```sh
git stash
git checkout v2.0.2
```
or you can just save them off in a branch of your own
```sh
git checkout -b mychanges-`date +%s`
git add ..
git commit -m "wip"
git checkout v2.0.2
```

### Audit output I get on the Block Explorer

```sh
~/btc-rpc-explorer $ npm audit
                                                                                
                       === npm audit security report ===                        
                                                                                
# Run  npm install meow@7.0.1  to resolve 1 vulnerability
SEMVER WARNING: Recommended action is a potentially breaking change
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Low           │ Prototype Pollution                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ yargs-parser                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ meow                                                         │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ meow > yargs-parser                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1500                            │
└───────────────┴──────────────────────────────────────────────────────────────┘

# Run  npm update lodash --depth 6  to resolve 4 vulnerabilities
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Low           │ Prototype Pollution                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ lodash                                                       │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ bitcoin-core                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ bitcoin-core > lodash                                        │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1523                            │
└───────────────┴──────────────────────────────────────────────────────────────┘

[... 18 lines cut ...]
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ pug > pug-code-gen > pug-attrs > constantinople >            │
│               │ babel-types > lodash                                         │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1523                            │
└───────────────┴──────────────────────────────────────────────────────────────┘

┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Low           │ Prototype Pollution                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ lodash                                                       │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ pug                                                          │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ pug > pug-filters > constantinople > babel-types > lodash    │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1523                            │
└───────────────┴──────────────────────────────────────────────────────────────┘

found 5 low severity vulnerabilities in 392 scanned packages
  run `npm audit fix` to fix 4 of them.
  1 vulnerability requires semver-major dependency updates.
```
Do you get the same? I'm not overly concerned about these
Addressing these should go back to that project https://github.com/janoside/btc-rpc-explorer

lucasmoten, 2020-07-17:
> Prior to starting a Raspibolt guide, you use electrum over TOR. SO your proxy tab is enabled with Use proxy and Use Tor.
> Now in latest electrum version it's just Use proxy check.
> So when you go [...]

Yes. This does make sense. I ran into this same interface change when I updated Electrum on my client as well.  There is value added in giving guidance

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus