raspibolt/raspibolt #639: Doubt when following the electrs tutorial for electrum server

jonjonez, 2020-07-26:
Following the tutorial for the electrum server https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html , I show the message in the tutorial below:

🚨 Change the password, otherwise Electrs is not able to talk to Bitcoin Core.

Change which password?

Password for the RPC or my bitcoin core wallet?

And if it is the RPC password, should I change the password in this electrs.conf file or in bitcoin.conf too?

That part was not very clear.

Thank you in advance for anyone willing to help.

Stadicus, 2020-07-27:
Thanks, will make that more specific. The idea is to simply replace `[PASSWORD_B]` to the `rpcpassword` already configured for Bitcon Core.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus