raspibolt/raspibolt #642: Raspi broken - no SSH connection :(

waffi121, 2020-08-09:
Hello Together

Today I tried to connect to my Raspi via SSH. Unfortunately I can't get a connection anymore, I tried the SSD card on another Raspberry, but it doesn't work either :(

I have the card on my windows computer at the moment and can save the lightning data via linux reader.

my question, what do i have to save so i can restore it later? 

i think the best thing is to backup the most important data and then reinstall everything again and restore it with the backed up data? 

thanks for the help and have a nice sunday :)

greetings waffi

lucasmoten, 2020-08-16:
I'm assuming that the Pi powers up just fine?

It may be that your IP address changed for your Pi.  This can happen if your network is auto assigning IP addresses via DHCP and you didn't reserve an IP for that device by mac address association.  Consider using tools to identify the new IP address of the pi and SSH to that
- https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html#find-it
- https://www.raspberrypi.org/documentation/remote-access/ip-address.md

Have you tried connecting the Pi to a monitor and logging in directly with keyboard?

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus