raspibolt/raspibolt #646: Prefer ElectrumX to Electrs

craigraw, 2020-09-17:
This guide used to suggest installing ElectrumX instead of Electrs as the Electrum Server. However, particularly for a RPi installation, Electrs is a poor choice. 

Due to an inefficient (and buggy) script hash subscription (see multiple open Electrs issues for details, but primarily https://github.com/romanz/electrs/issues/146) implementation, on an RPi4 Electrs tends to use 100% of one of the CPU cores when it is performing a PeriodicUpdate scan with just 100 script hashes subscribed. With the default ``wait_duration_secs`` setting of 5 seconds, this often means the server is permanently locked up in this work. The implementation may work on a laptop but on my (RPi4, 8GB RAM) node it results in periodic disconnections and frequent timeouts doing basic history retrieval operations. It should also be noted that Electrs often fails to notify when a script hash is updated. Increasing ``wait_duration_secs`` improves the situation only marginally.

ElectrumX on the other hand is efficient, supporting batched JSON-RPC calls and near instant responses regardless of the number of script hashes subscribed. This guide already recommends installing bitcoind with txindex=1, so using Electrs (which does not require this setting) provides no advantage here. ElectrumX is also marginally smaller in disk space usage, and is much lighter in CPU usage. Note that ElectrumX has been [forked](https://github.com/spesmilo/electrumx) by the Electrum maintainers to continue supporting the BTC protocol.

In short ElectrumX is a much better implementation that is lighter on the system in both CPU and disk usage, and I believe much better suited to the hardware this guide is written for.

Pantamis, 2020-09-22:
[Relevent benchmark](https://blog.keys.casa/electrum-server-performance-report/)

I can't have more than two instances of Electrum served by ElectRS it is a bit boring so I am quite interested by this improvment.

Do you have a guide I can follow to install ElectrumX on the RaspiBolt ?

Initial sync seems to be a BIG issue with ElectrumX as it crashes many time. How did you handle that (did you sync on another device) ? Or is it solved already ?

Thanks for the feedback !

craigraw, 2020-09-23:
I actually used a previous version of the Raspibolt guide to install ElectrumX some time ago. I can't remember any significant issues syncing initially, but I may have experienced some issues that I've since forgotten about. Are you running it on an SSD?

What I can say is that, apart from the compact database script which needs to be run every year or so, ElectrumX has been remarkably stable and handled everything I've thrown at it graciously.

I also find I can't open more than one wallet on an Electrs installation without it seizing up.

Pantamis, 2020-10-08:
Ok so I tried to run ElectrumX on the raspberry it is quite messy. Feedback:

I had to sync the database on my laptop and call my node remotely from my LAN and it tooks me almost two weeks to find the good setup for CACHE_MB variable so that it is not slower and complete the sync. This thing is quite annoying, I had to stop and restart the sync of the database from my laptop because for reasons (LevelDB caching ?) I don't know, the sync becomes very slow once the cache has been filled and empty once or twice (rebooting the sync is much more efficient than flushing, why ?)

Once my database is at the tip, everything works fine on my laptop.

But on the Raspberry... Oh boy...

I copy-pasted the database with scp in the Pi and launch ElectrumX. The time to copy (1H30-2H for 50-60 GB), ElectrumX is not in sync anymore. So it tries to catch the tip after opening the database. But the Prefetcher is hanging at the following logs line:

 ` INFO:Prefetcher:catching up to daemon height XXXX (XXX blocks behind)`

One hour later, still blocked. It takes just too much time. When I stop ElectrumX (because I had to set other Env variable) it doesn't flush the DB and it is still hanging at

`INFO:BlockProcessor:flushing to DB for a clean shutdown...`

and it is killed by systemd after 30min (I set the timeout to 45min now, but didn't try to see if it has enough time to stop.
Those durations are too high to stay in sync if ElectrumX is started or stopped, do you experiment such long time when you start and stop your ElectrumX server ?

 It looks like a bad idea to leave Electrs for ElectrumX, unless you are able to run ElectrumX on your Raspberry Pi (with LND, Bitcoin Core and BlockExplorer running) and have some magic value for the configuration file. If you run with a SSD and put the DB in it, that may be the reason why you are able to stop and restart ElectrumX while staying in sync. It is not a hardware requirement of RaspiBolt so it is worth the warning....

craigraw, 2020-10-08:
Thanks for the detailed investigation. I was sure I had installed ElectrumX from a previous Raspibolt guide - but I guess not!

I think the key here as you suggest is an SSD drive. ElectrumX is almost certainly I/O bound on an HDD. I see that an SSD is recommended but not required for Raspibolt, and I understand the desire to keep hardware costs down. My setup is an SSD so that explains the difference in experience. It's worth repeating though that with an SSD, ElectrumX is much lighter in system resources than Electrs, particularly CPU-wise (I have run both on SSD). With the default `wait_duration_secs` setting on Electrs one core on the RPi4 is busy almost all the time given a hundred script hash subscriptions (two small wallets).

On the other side, I have filed a number of issues on the Electrs project: https://github.com/romanz/electrs/issues?q=is%3Aopen+commenter%3Acraigraw

Hopefully they will get some attention!

Pantamis, 2020-10-08:
Ok I can always reliably catch the chaintip but it is veeery long. 

It looks indeed plausible that the limit comes from the I/O with HDD so you may not experiment them thanks to the SSD. Most of the time, ElectrumX does nothing on my Pi (no CPU usage, and not much IO), when syncing in particular. Once sync is completed, it seems also that ElectrumX is doing too much requests to the Bitcoin node, ElectrumX logs are full of 
`ERROR:Daemon:daemon service refused: Work queue depth exceeded.  Retrying occasionally...`
and
`INFO:ElectrumX: incoming request Request('blockchain.estimatefee', [50]) timed out after 30 secs`

I test opening several wallets from client, it works but it is quite slow and not very robust to me.
For example, the blockexplorer cannot work with it because it makes too much request (at least, ElectRS cannot have this problem since it didn't use the txindex of Core so less RPC calls). And everything else seems slow.

So I think we should propose ElectrumX only with SSD for the RaspiBolt, not recommended otherwize. Also, I used LevelDB for ElectrumX, maybe RocksDB is better ? (it would surprise me)

I think using ElectRS is still a better idea for a raspberry pi anyway. Solving the issues you raise in ElectRS is maybe the only way to have decent perf with several wallets open.

Thanks for your work on this ! (Sparrow is great ! Once issues with ElectRS are solved, it will be amazing !)

craigraw, 2020-10-09:
> So I think we should propose ElectrumX only with SSD for the RaspiBolt, not recommended otherwize.

That makes sense. I do think it's a worthwhile upgrade for some power users. I know I love my ElectrumX instance!

> I think using ElectRS is still a better idea for a raspberry pi anyway. Solving the issues you raise in ElectRS is maybe the only way to have decent perf with several wallets open.

I think it depends on the hard drive - but given how popular Electrs is I really hope the issues are resolved soon.

> Thanks for your work on this ! (Sparrow is great ! Once issues with ElectRS are solved, it will be amazing !)

Thanks! Love the Raspibolt guide too, always my goto reference when installing or upgrading nodes.

Pantamis, 2020-10-13:
I bought a SSD to try on RaspiBolt. Everything is crazy fast when using it (any sync can be done in one day) but I ran into trouble when I plug it on the Pi.

My setup was the following: 
- HDD 2TB to store blocks only on port USB 3.0
- SSD 500GB to store indexes (blocks/txndex/electrumx/lnd) on port USB 3.0

The problem is that the Raspberry Pi is not powerful enough to power up the SSD and the HDD at the same time. It seems that the SSD can have very high peak of power consumption, the Pi disconnects every USB since it cannot provide more than 1.2A for all USB devices. Did you have this problem ? Are you using only a SSD and no HDD ? How did you solve it otherwise ?

Appart from this, I think it is clearly worth it to add a medium sized SSD, the UX is definitively better. Big SSDs are to expansive and small are too small so it would be really nice to have a setup which works with both SSD and HDD.

Pantamis, 2020-10-13:
I don't know if its the HDD or SSD but `usb-devices` shows `MxPwr=900mA` for the SSD and 0 for HDD. But maybe the current intensity spikes a lot when the HDD's disk starting spinning.

I found people using successfully 2 SSD at the same time with a Raspberry Pi. So you may be right.

Sad we can't have this medium-sized medium-budget setup. I think it is a lot more better to use old laptops than a Raspberry for that. I will see if some succeed with a externally powered USB hub.

craigraw, 2020-10-15:
Great news! It's really hard to go back to Electrs after ElectrumX, but I've spent the last week tuning Sparrow's management of requests to Electrs and it's now considerably better, to the extent that everything still works (albeit slowly) even when Electrs is running at 100% CPU. I'm also chatting to Roman the developer about some of the issues.

Pantamis, 2020-10-16:
Nice !
I hope you can make it :p, ElectRS is still a good compromise for cheap hardware, Rust is doing the job I guess (it is incredibly fast at syncing)

craigraw, 2021-06-24:
Yes, I've been testing Electrs 0.9.0 and it's performance, while not quite at ElectrumX levels, is very much better, particularly when using batched JSON-RPC. I'll close this off.

PS Thanks for the kind words on Sparrow!

closed 2021-06-24 by craigraw