raspibolt/raspibolt #648: Notes on updating to 0.11.1

tdrusk, 2020-10-09:
Here's the procedure I followed to update to 0.11.1.  Hope it helps update the documentation. There is a recent vulnerability which suggests upgrading to this new version. https://groups.google.com/a/lightning.engineering/g/lnd/c/NtG33wYrHdo/m/fWam0XbLAQAJ

```
$ cd /tmp
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.11.1-beta/lnd-linux-armv7-v0.11.1-beta.tar.gz
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.11.1-beta/manifest-v0.11.1-beta.txt
$ wget https://github.com/lightningnetwork/lnd/releases/download/v0.11.1-beta/manifest-v0.11.1-beta.txt.sig
$ wget https://keybase.io/bitconner/pgp_keys.asc

$ sha256sum --check manifest-v0.11.1-beta.txt --ignore-missing
lnd-linux-armv7-v0.11.1-beta.tar.gz: OK

$ gpg ./pgp_keys.asc
> gpg: WARNING: no command supplied.  Trying to guess what you mean ...
> pub   rsa4096 2018-03-06 [SC] [expires: 2021-03-06]
>      9C8D61868A7C492003B2744EE7D737B67FA592C7
> uid           Conner Fromknecht <conner@lightning.engineering>
> sub   rsa4096 2018-03-06 [E] [expires: 2021-03-06]

$ gpg --import ./pgp_keys.asc
$ gpg --verify manifest-v0.11.1-beta.txt.sig
> gpg: assuming signed data in 'manifest-v0.11.1-beta.txt'
> gpg: Signature made Fri 02 Oct 2020 00:38:32 BST
> gpg:                using RSA key 9C8D61868A7C492003B2744EE7D737B67FA592C7
> gpg: Good signature from "Conner Fromknecht <conner@lightning.engineering>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 9C8D 6186 8A7C 4920 03B2  744E E7D7 37B6 7FA5 92C7

$ tar -xzf lnd-linux-armv7-v0.11.1-beta.tar.gz
$ sudo install -m 0755 -o root -g root -t /usr/local/bin lnd-linux-armv7-v0.11.1-beta/*
$ lnd --version
> lnd version 0.11.1-beta commit=v0.11.1-beta
```

robot-dreams, 2020-10-09:
Thanks!

This is helpful, especially because the instructions at https://stadicus.github.io/RaspiBolt/raspibolt_40_lnd.html#download refer to the pgp_keys.asc of `roasbeef`, which just expired last week on 10/2/2020.

lucasmoten, 2020-10-13:
Good catch. Can you open a PR with that update?

jambolo, 2020-11-14:
I opened #657, which fixes some typos introduced in the previous commit and gets roasbeef's old key from a key server.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus