raspibolt/raspibolt #651: Create a mempool.space block explorer option to run as block explorer

raelschiffler, 2020-10-12:
Sorry if this is outside the project/section scope, it's not even an issue, just a suggestion, but I like using mempool.space because it has a really cool design, features and layout and their project is also open source on github: https://github.com/mempool/mempool
You have a very didactic way to teach and it would be nice if you could add to the Block Explorer Section a guide to install mempool.space. Thanks a lot. I'm very thankful to you for teaching me how to create a full node on a Raspberry and it's all working fine. I've also tipped you.

lucasmoten, 2020-10-13:
I'm looking into this following the guidance for the v1 version outlined here https://github.com/mempool/mempool#mempool-v1-not-using-docker-advanced. If I come up with anything I'll post it back here first.  If anyone else is working on adding mempool to raspibolt, please discuss here.

Stadicus, 2021-11-27:
A bonus guide for mempool.space would be cool indeed :)

raelschiffler, 2022-01-26:
Don't let it die.

VajraOfIndra, 2022-01-27:
> Don't let it die.

Did you have a look at #870 ? We'd appreciate if you give the guide a try and send any feedback that could improve it!

VajraOfIndra, 2022-02-17:
> Don't let it die.

@raelschiffler the Mempool bonus guide is [now published](https://raspibolt.org/bonus/bitcoin/mempool.html). For now it works within the LAN only, we need to add a remote connection via Tor. Feel free to propose ideas or PRs to improve the Mempool guide.

Can we close this Issue?

raelschiffler, 2022-02-17:
> > Don't let it die.
> 
> @raelschiffler the Mempool bonus guide is [now published](https://raspibolt.org/bonus/bitcoin/mempool.html). For now it works within the LAN only, we need to add a remote connection via Tor. Feel free to propose ideas or PRs to improve the Mempool guide.
> 
> Can we close this Issue?

Yes, sure.

closed 2022-02-17 by raelschiffler

raelschiffler, 2022-04-15:
> > Don't let it die.
> 
> @raelschiffler the Mempool bonus guide is [now published](https://raspibolt.org/bonus/bitcoin/mempool.html). For now it works within the LAN only, we need to add a remote connection via Tor. Feel free to propose ideas or PRs to improve the Mempool guide.
> 
> Can we close this Issue?

Hi, @VajraOfIndra. The link you provided is broken. Could you check please?

VajraOfIndra, 2022-04-15:
> Hi, @VajraOfIndra. The link you provided is broken. Could you check please?

Yes, the link I provided was created before the big structural change we made in the repository. That change broke most external links to the guide unfortunately.

The Mempool bonus guide is now at: https://raspibolt.org/guide/bonus/bitcoin/mempool.html