raspibolt/raspibolt #652: Notes on Adding Ride the Lightning and Remote Access

lucasmoten, 2020-10-12:
I plan to work this into another Bonus Page at some point, but for anyone looking for a guide on setting up Ride the Lightning on RaspiBolt, the following should be a good starting point. It's based in part on the https://github.com/Ride-The-Lightning/RTL guidance, as well as the structure for [Block Explorer](https://github.com/Stadicus/RaspiBolt/blob/master/raspibolt_55_explorer.md), setting up as another user, and borrowinng from the NGINX setup done for Electrs for secure communication.  

----

Ride the Lightning notes

## Preparations

### Install NodeJS

- Starting with user “admin”, we switch to user “root” and add the Node JS package repository. We’ll use version 12 which is the most recent stable one.  If you installed BTC RPC Explorer, then you've already accomplisehd this step. When finished, exit the “root” user session.

```sh
$ sudo su
$ curl -sL https://deb.nodesource.com/setup_12.x | bash -
$ exit
```

```
Install NodeJS using the apt package manager.

$ sudo apt-get install nodejs
$ exit
```

### Firewall

- Configure firewall to allow incoming HTTP requests from your local network to the web server.

```sh
$ sudo ufw allow from 192.168.0.0/16 to any port 3000 comment 'allow Ride the Lightning from local network'
$ sudo ufw allow 3001 comment 'allow Ride the Lightning'
$ sudo ufw status
```

## Ride the Lightning

### Installation

We do not want to run Ride the Lightning alongside bitcoind and lnd because of security reasons. For that we will create a separate user and we will be running the code as the new user. We are going to install Ride the Lightning in the home directory since it doesn’t take much space and doesn’t use a database.

Create a new user with your password [ A ] and open a new session
```sh
$ sudo adduser ridethelightning
$ sudo cp /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/ridethelightning/admin.macaroon
$ sudo chown ridethelightning:ridethelightning /home/ridethelightning/admin.macaroon
$ sudo su - ridethelightning
```

Download the release from GitHub and install dependencies

First, we start with the keys used to sign the release file

```
$ wget https://keybase.io/suheb/pgp_keys.asc
$ gpg ./pgp_keys.asc

gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2019-09-14 [SC] [expires: 2022-09-27]
      3E9BD4436C288039CA827A9200C9E2BC2E45666F
uid           saubyk (added uid) <39208279+saubyk@users.noreply.github.com>
uid           Suheb <39208279+saubyk@users.noreply.github.com>
sub   rsa4096 2019-09-14 [E] [expires: 2022-09-27]

$ gpg --import pgp_keys.asc
```

The output should appear as follows

```
gpg: key 00C9E2BC2E45666F: public key "saubyk (added uid) <39208279+saubyk@users.noreply.github.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
```

Continue with downloading the release file and then checking signature

```sh
$ gpg rm pgp_keys.asc
$ wget https://github.com/Ride-The-Lightning/RTL/archive/v0.9.0.tar.gz
$ wget https://github.com/Ride-The-Lightning/RTL/releases/download/v0.9.0/v0.9.0.tar.gz.asc
$ gpg --verify v0.9.0.tar.gz.asc v0.9.0.tar.gz
```

Expected output

```
gpg: Signature made Wed 26 Aug 2020 03:57:33 PM BST
gpg:                using RSA key 3E9BD4436C288039CA827A9200C9E2BC2E45666F
gpg: Good signature from "saubyk (added uid) <39208279+saubyk@users.noreply.github.com>" [unknown]
gpg:                 aka "Suheb <39208279+saubyk@users.noreply.github.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 3E9B D443 6C28 8039 CA82  7A92 00C9 E2BC 2E45 666F
```

Extract the files from the compressed file and install

```sh
$ tar -xvf v0.9.0.tar.gz
$ cd RTL-0.9.0
$ npm install --only=prod
```

Several lines of compiling will be displayed and the end should look similar to the following

```
Compiling ngx-perfect-scrollbar : main as umd
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 236 packages from 235 contributors and audited 1555 packages in 743.292s

4 packages are looking for funding
  run `npm fund` for details

found 1 high severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
```

### Configuration

We modify the sample configuration file and tailor to our needs

```sh
mv sample-RTL-Config.json RTL-Config.json
nano RTL-Config.json
```

- Near the top of the file you should set the password. 

```json
  "multiPass": "password"
```

In the nodes array, set the values for macaroonPath and configPath

- Set the macaroonPath

```json
        "macaroonPath": "/home/ridethelightning"
```

- Set the configPath

```json
        "configPath": "/home/bitcoin/.lnd/lnd.conf"
```
- Save and exit

### First start

Test starting Ride the Lightning manually first to make sure it works.

```sh
$ cd ~/RTL-0.9.0
$ node rtl
```

If the server started successfully, you should get the below output on the console:
```
Server is up and running, please open the UI at http://localhost:3000
```

Now point your browser to http://raspibolt.local:3000 (or whatever you chose as hostname) or the ip address (e.g. http://192.168.0.20:3000). You should see the home page of the Ride the Lighnting prompting you to login.

*** IMAGE 1
[image]

Specify the password you set for multiPass above and login.  You should see your balances, liquidity and options to receive and pay invoices.

**** IMAGE 2
[image]

Logout

- Stop the Ride the Lightning in the terminal with `CTRL`-`C` and exit the "ridethelightning" user session

```sh
$ exit
```

### Autostart on boot

Now we'll make sure our block explorer starts as a service on the Raspberry Pi so it's always running.
In order to do that we create a systemd unit that starts the service on boot directly after Bitcoin Core.

- As user "admin", create the service file.

```sh
$ sudo nano /etc/systemd/system/ridethelightning.service
```

- Paste the following configuration. Save and exit.

```
# RaspiBolt: systemd unit for Ride the Lightning
# /etc/systemd/system/ridethelightning.service

[Unit]
Description=Ride the Lightning
After=network.target bitcoind.service
After=lnd.service

[Service]
WorkingDirectory=/home/ridethelightning/RTL-0.9.0
ExecStart=/usr/bin/node rtl
User=ridethelightning

# Restart on failure but no more than 2 time every 10 minutes (600 seconds). Otherwise stop
Restart=on-failure
StartLimitIntervalSec=600
StartLimitBurst=2

[Install]
WantedBy=multi-user.target
```

- Enable the service, start it and check log logging output.

```sh
$ sudo systemctl enable ridethelightning.service
$ sudo systemctl start ridethelightning.service
$ sudo systemctl journalctl -f -u ridethelightning.service
```

- You can now access your own Ride the Lightning from within your local network by browsing to http://raspibolt.local:3000 (or your equivalent ip address).

### Secure communication

We should only communicate with Ride the Lightning over an encrypted channel.  We will use NGINX the same way we did for Electrum, and this section of the guide assumes you've already installed NGINX, created the self-signed TLS certificate, and configured nginx.conf for Electrs.  We continue with modifying the configuration for Ride the Lightning

```sh
$ sudo nano /etc/nginx/nginx.conf
```

Add an additional upstream
```
  upstream rtl {
    server 127.0.0.1:3000;
  }
```

And add an additional server
```
  server {
    listen 3001 ssl;
    proxy_pass rtl;
  }
```

Save and exit the file.  Test the NGINX continuration and restart the service

```sh
$ sudo nginx -t
$ sudo systemctl restart nginx
```

### Remote access over Tor (optional)

You can easily add a Tor hidden service on the RaspiBolt and access the Ride the Lightning interface with the Tor browser from any device.

- Add the following three lines in the section for "location-hidden services" in the `torrc` file. Save and exit

```sh
$ sudo nano /etc/tor/torrc
```

```
HiddenServiceDir /var/lib/tor/hidden_service_rtl/
HiddenServiceVersion 3
HiddenServicePort 3001 127.0.0.1:3001
```

Restart Tor and get your connection address.

```sh
$ sudo systemctl restart tor
$ sudo cat /var/lib/tor/hidden_service_rtl/hostname
> abcefg...................zyz.onion
```

With the Tor browser (link this), you can access this onion address from any device.  

**Congratulations!** You now have Ride the Lightning running to manage your Lightning service on our own node.

lucasmoten, 2020-10-13:
https://github.com/Ride-The-Lightning/RTL/tree/v0.9.1 is available at the time of this writing. I plan to check out the beta release of that to work into this guide.

lucasmoten, 2020-10-13:
Some TODO items for improvement
- Clarify that the external/remote port for access over Tor is on port 3001 with https, which will proxy through nginx to local service.
- Consider either extract to temp folder and move to generic /RTL without version, and giving guidance on doing upgrades.  Note that the systemd service definition bakes in RTL-0.9.0 and would need to be changed in this as well
- For common sections reused from other services (NodeJS, NGINX), maybe the base installation of those should be in their own pages? 
- Would be nice to add to the System Overview with Tor link

bmml1, 2020-10-22:
In LND Upgrade section one must delete all macaroons before an upgrade.

> Remove old stuff sudo rm /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*.macaroon, then download, verify and install the latest LND binaries as described in the Lightning section of this guide.

I don't know if this is actually necessary?
However if following this step we also need to do this again after upgrading LND: 

```
$ sudo adduser ridethelightning
$ sudo cp /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/ridethelightning/admin.macaroon
$ sudo chown ridethelightning:ridethelightning /home/ridethelightning/admin.macaroon
$ sudo su - ridethelightning
```
We should mention this somewhere. Or did I get something wrong?

Cheers

lucasmoten, 2020-11-05:
Good catch. I copy/pasta'd and it shows ;) i'll go back and clean up

Longmont, 2020-12-20:
[image]
When I attempt to manually start the RTL it fails. I don't know enough about JSON, to know what needs fixed. It appears the problem is in the connect.js file. What do the numbers 552:11 mean? And something is incorrect with the Object. anonymous line as well. Did I create/download the files in the wrong user directory?

Longmont, 2020-12-23:
> [image]
> When I attempt to manually start the RTL it fails. I don't know enough about JSON, to know what needs fixed. It appears the problem is in the connect.js file. What do the numbers 552:11 [...]

Figured out the problem. I put the wrong path for the macaroon. I was pointing it back to the path under the bitcoin user. Noob problems. Issue resolved.

lucasmoten, 2021-04-19:
- Updated to v0.10.1
- Changed to use /home/ridethelightning/RTL as common install path instead of versioned
- Added guidance on upgrading

martinatime, 2021-05-13:
@lucasmoten Awesome write up! I got it working on my first attempt. Thanks!

VajraOfIndra, 2021-09-05:
@lucasmoten do you have plan to make a PR to add this awesome guide as an optional Raspibolt guide? It would be awesome to see it merge.

Stadicus, 2021-11-14:
Awesome, thanks for your writeup! I rewrote it a bit to make it potentially an integral part of the RaspiBolt v3:   
https://stadicus.github.io/raspibolt-remote/raspibolt_57_rtl.html

Stadicus, 2021-11-27:
Ride The Lighnting is now part of the RaspiBolt v3 core guide! :tada: 

https://raspibolt.org/rtl.html

closed 2021-11-27 by Stadicus