raspibolt/raspibolt #653: Sync speed

gregkbarnes, 2020-10-20:
I am running on a Raspberry PI 4 (4GB). I've setup following the guide up until point in which I need to sync.

After ~24hrs, my blockchain progress is < 0.2. I have made sure my HDD is setup to run efficiently and it was clocked at >100MB/s. I have set the cache to 2000MiB

However, the speed is just crawling. I am thinking it may be due to TOR, as I have synced the blockchain from scratch earlier this year (clearnet) - I saw much faster results.

Any ideas?

lucasmoten, 2020-10-20:
When I last synched on Raspberry Pi 4 with 4GB, it took me a little less than a week on SSD over wifi, a few days with SSD on ethernet, and over 30 days on a non SSD drive.  

I believe that progress of .2 is 20% not .2% so its possible you just have a few days to go.  What block number and/or date is it up to?

If it's not SSD, and you dont have time to waste, then I'd stop the process now, and get an SSD.  Outside of extreme budgetary concerns, theres no reason to not use SSD or better these days.  The ROI will even pay for itself over time from the differential in power demand

Pantamis, 2020-10-21:
Can confirm this too. I made the sync on my laptop and plug the HDD after. (Now I have an SSD for chainstate and an HDD for blocks, it gives resilience in case you need to reindex because of power outage.)

@lucasmoten : "Outside of extreme budgetary concerns, theres no reason to not use SSD or better these days."
Is there any issue running bitcoin which uses levelDB, an key-val store optimized for HDD, with a SSD ? I often read that the issue is that SSDs have a limited number of write on the same "sector" so they are degraded quite fast by sequential read/write optimization of levelDB. If that's a myth now (I also read that SSD "distribute randomly" where they write to uniformize usage), it is worth to indicate in RaspiBolt's hardware requirement that an SSD can be used without worrying for better performance and UX. That way, people don't feel "blocked to change for SSD" by the HDD they bought for the node.

gregkbarnes, 2020-10-21:
Gents - thanks very much for the help. As it turns out, I'm currently on block 464708 :) So well on my way there.

As far as my choice to go HDD vs SDD, it was simply the only 'extra' drive I had laying around with adequate storage. The extra SSDs I have are only 250 GB.

{
  "chain": "main",
  "blocks": 464708,
  "headers": 653752,
  "bestblockhash": "000000000000000000b86ad5da83ce28edb5190365d4b4385610b5118fd7                                                                                          1d1c",
  "difficulty": 521974519553.6282,
  "mediantime": 1493851508,
  "verificationprogress": 0.3803922548538838,
  "initialblockdownload": true,
  "chainwork": "0000000000000000000000000000000000000000005307bf17c2a5d7c2a32314                                                                                          ",

closed 2020-10-21 by gregkbarnes