raspibolt/raspibolt #655: Installing LiT instead of LND (LND + Loop (+ GUI) + Faraday + Pool all at once)

Pantamis, 2020-11-04:
[Lightning terminal](https://github.com/lightninglabs/lightning-terminal) is a project of Lightning Labs the main feature is that it is a package of softwares, it contains the last release of LND, Loop, Faraday and the new Pool market client. It also has a nice GUI which will support more and more features (only Loop for now, but it is obvious that there will be a faraday and pool GUI, and maybe some function like channel rebalancing).

The installation process is exactly the same as LND so it essentialy gives new feature to RaspiBolt user for free. The only condition to use them is to add alias for the commands `loop frcli pool` to tell them to send gRPC to port 10009 instead of default ones as described [here](https://github.com/lightninglabs/lightning-terminal/blob/master/doc/config-lnd-integrated.md#example-loop-command)

It is still in alpha for now, but it could be nice to install LiT instead of LND, or at least propose it as an option in the guide at the Lightning section.

Can make a PR to do that, what do you thing ?

argentoblitz, 2020-11-21:
Would love this. Trying to figure out getting Pool working on my bolt and failing. Would really help with making my part of the Lightning network feel less alone and underutilized. I have liquidity to offer, but right now perhaps in all the wrong places.

Pantamis, 2020-11-23:
Ok I made a guide to install LiT in the same section as Lightning but I have trouble to create a pull request (I already have one on WireGuard installation which is waiting for review and I sadly made it in my master branch)

You can still take alook at it here: https://github.com/Pantamis/RaspiBolt/blob/master/raspibolt_40_lnd.md

Tell me if it works or not.

If it is worth it then I will fork the repo once again to propose this in a clean seperated PR before the merge of my WireGuard PR.

penguin4111, 2021-02-17:
Thanks for the guide, I tried it out. I only had a couple of issues:

1. You write "mkdir /home/bitcoin/.lnd" but I think you  mean "mkdir /home/bitcoin/.lit" since that is the location where litd is looking for the .conf file by default.
2. Connecting to the UI on my browser did not work until I added the line "httpslisten=0.0.0.0:8443" to the beginning of the .conf file.
3. Worth noting that the ui password must be at least 8 characters long or it will give an error.

Pantamis, 2021-02-17:
Thanks a lot for this great feedback ! I am glad it mainly worked for you  :)

Will do the diff tomorrow (along with updates of URL to download the last binaries with Pool UI !)

VajraOfIndra, 2021-09-05:
This issue could be closed as pull request #738 merged the LiT installation guide to the main guide :)

Pantamis, 2021-09-06:
Indeed !

I still have to modify the LiT guide to use it with LND in remote mode (not much work but I lack of time)

closed 2021-09-06 by Pantamis