raspibolt/raspibolt #656: Installing / updating Tor fails

bmml1, 2020-11-04:
Hi,

updating and I guess installing tor to does not work anymore with the sources used in the privacy section:

`deb https://deb.torproject.org/torproject.org buster main`
`deb-src https://deb.torproject.org/torproject.org buster main`

`sudo apt update `

results in:

`N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://deb.torproject.org/torproject.org buster InRelease' doesn't support architecture 'armhf'
`

What source has to be used now?

pretyflaco, 2020-11-08:
I have the same issue.

openoms, 2020-11-10:
Listed some possible solutions here: https://github.com/rootzoll/raspiblitz/issues/1752#issue-739776875

But first of all make sure your ISP (or router settings / parenteral control) is not blocking torproject.org in your network.

Also see: https://t.me/RaspiBlitzFAQ/156
```
How to use #nslookup to examine DNS service censorship:

sudo apt-get install dnsutils -y

nslookup https://deb.torproject.org/torproject.org
Server:    10.168.0.1
Address:  10.168.0.1#53

Non-authoritative answer:
Name:  https://deb.torproject.org/torproject.org
Address: 18.130.115.151
Name:  https://deb.torproject.org/torproject.org
Address: ::1

Run OONI to build the community database about restrictive ISP-s: https://ooni.org/install/
```

2pac1, 2020-12-17:
Try this:
sudo nano /etc/apt/sources.list

add this:
deb     [arch=amd64] https://deb.torproject.org/torproject.org buster main
deb-src [arch=amd64] https://deb.torproject.org/torproject.org buster main

per: https://support.torproject.org/apt/tor-deb-repo/

I have the same issue. With this: apt update shows:
1 package can be upgraded. Run 'apt list --upgradable' to see it.

And you cant upgrade it.

Covalent5, 2020-12-23:
Same issue here. Any news on this matter? I am stuck with the fallback version: 0.3.5.12.

```
admin ~ ₿ apt list --upgradable -a
Listing... Done
tor-geoipdb/stable 0.4.4.6-1~d10.buster+1 all [upgradable from: 0.3.5.12-1]
tor-geoipdb/stable,now 0.3.5.12-1 all [installed,upgradable to: 0.4.4.6-1~d10.buster+1]
```

```
admin ~ ₿ sudo apt-get install tor-geoipdb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 tor-geoipdb : Depends: tor (>= 0.4.4.6-1~d10.buster+1) but 0.3.5.12-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

2pac1, 2020-12-23:
It does seem that that version was released just recently:
https://gitweb.torproject.org/tor.git/plain/ChangeLog?h=tor-0.3.5.12

2pac1, 2020-12-23:
Also,.can't remember 100% but I'm 64bit beta I think you end up on the same
version. I forgot as i did too many tests.

Covalent5, 2020-12-23:
> It does seem that that version was released just recently:
> https://gitweb.torproject.org/tor.git/plain/ChangeLog?h=tor-0.3.5.12
> 
> 
> > Same issue here. Any news on this matter? I am stuck with the fallback
> > version: 0.3.5.12.
> >
> > admin ~ ₿ apt list --upgradable -a
> >
> > Listing... Done
> >
> > tor-geoipdb/stable 0.4.4.6-1~d10.buster+1 all [upgradable from: 0.3.5.12-1]
> >
> > tor-geoipdb/stable,now 0.3.5.12-1 all [installed,upgradable to: 0.4.4.6-1~d10.buster+1]
> >
> >
> > admin ~ ₿ sudo apt-get install tor-geoipdb
> >
> > Reading package lists... Done
> >
> > Building dependency tree
> >
> > Reading state information... Done
> >
> > Some packages could not be installed. This may mean that you have
> >
> > requested an impossible situation or if you are using the unstable
> >
> > distribution that some required packages have not yet been created
> >
> > or been moved out of Incoming.
> >
> > The following information may help to resolve the situation:
> >
> >
> >
> > The following packages have unmet dependencies:
> >
> >  tor-geoipdb : Depends: tor (>= 0.4.4.6-1~d10.buster+1) but 0.3.5.12-1 is to be installed
> >
> > E: Unable to correct problems, you have held broken packages.

Then why does it list as an upgradable?

2pac1, 2020-12-23:
Don't know, maybe they reverted back to this "old" version.

Someone with more info might contribute.

openoms, 2021-01-06:
See the solutions in: https://github.com/rootzoll/raspiblitz/issues/1906#issuecomment-755299759

dominicus, 2021-10-02:
https://gitlab.torproject.org/tpo/core/tor/-/issues/40347

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus