raspibolt/raspibolt #660: Wrong lines in installation guide

bvrulez, 2020-11-24:
In the installation guide [here](https://stadicus.github.io/RaspiBolt/raspibolt_40_lnd.html) you say

```
$ pgp ./roasbeef.asc
> 9769140D255C759B1EB77B46A96387A57CAAE94D
$ gpg ./bitconner.asc
> 9C8D61868A7C492003B2744EE7D737B67FA592C7
```

I get the error: 

```
admin@RPi4-2:/tmp $ pgp ./bitconner.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: no valid OpenPGP data found.
gpg: processing message failed: Unknown system error
WARNING: Can't find the right public key-- can't check signature integrity.

```

If I cat the file I get: 

```
admin@RPi4-2:/tmp $ cat roasbeef.asc
--2020-11-24 10:01:55--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 34.238.229.11, 34.238.91.171
Connecting to keybase.io (keybase.io)|34.238.229.11|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6900 (6.7K) [text/plain]
Saving to: ‘pgp_keys.asc’

     0K ......                                                100%  129M=0s

2020-11-24 10:01:56 (129 MB/s) - ‘pgp_keys.asc’ saved [6900/6900]

```
So this file seems to be just the output-log. I am not familiar with pgp. 

In the video below the text it does not show this part. It continues with: 

```
gpg ./pgp_keys.asc
gpg --import ./pgp_keys.asc 
```

So I suppose the text is incorrect. But since this seems to be an encryption thing i would like to sort it out.

bvrulez, 2020-11-24:
My EDIT seems to be correct. I get a different output for one of the keys, maybe it was renewed? 

```
admin@RPi4-2:/tmp $ pgp ./roasbeef.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2019-10-13 [C]
      E4D85299674B2D31FAA1892E372CBD7633C61696
uid           Olaoluwa Osuntokun <laolu32@gmail.com>
sub   rsa4096 2019-10-13 [E] [expires: 2024-10-11]
sub   rsa4096 2019-10-13 [S] [expires: 2024-10-11]
sub   rsa4096 2019-10-13 [A] [expires: 2024-10-11]
admin@RPi4-2:/tmp $ pgp ./bitconner.asc
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2018-03-06 [SC] [expires: 2021-03-06]
      9C8D61868A7C492003B2744EE7D737B67FA592C7
uid           Conner Fromknecht <conner@lightning.engineering>
sub   rsa4096 2018-03-06 [E] [expires: 2021-03-06]

```

The manual says:

```
$ pgp ./roasbeef.asc
> 9769140D255C759B1EB77B46A96387A57CAAE94D
$ gpg ./bitconner.asc
> 9C8D61868A7C492003B2744EE7D737B67FA592C7
```

bvrulez, 2020-11-24:
I thought this works. But another problem (?) comes up: 

```
admin@RPi4-2:/tmp $ gpg --verify roasbeef-manifest-v0.11.1-beta.txt.sig manifest-v0.11.1-beta.txt
gpg: Signature made Fri 02 Oct 2020 02:03:47 CEST
gpg:                using RSA key 4AB7F8DA6FAEBB3B70B1F903BC13F65E2DC84465
gpg: Can't check signature: No public key

```

Stadicus, 2020-11-24:
Quick fix the `wget` command in https://github.com/Stadicus/RaspiBolt/pull/661

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus