raspibolt/raspibolt #662: EPS fails on startup

OpenSystemCode, 2020-11-24:
Hi I am discovered the same Issue as https://github.com/Stadicus/RaspiBolt/issues/535

Bitcoind Version: Bitcoin Core version v0.20.0

Electrum Personal Server : electrum-personal-server 0.2.0

Config: https://stadicus.github.io/RaspiBolt/raspibolt_64_electrum.html

Output form `journaltctl`.

```
Nov 24 18:56:04 "" python3[10163]: INFO:2020-11-24 18:56:04,651: Starting Electrum Personal Server
Nov 24 18:56:04 "" python3[10163]: INFO:2020-11-24 18:56:04,652: Logging to /tmp/electrumpersonalserver.log
Nov 24 18:56:05 "" python3[10163]: INFO:2020-11-24 18:56:05,199: Displaying first 3 addresses of each master public key:
Nov 24 18:56:05 "" python3[10163]: INFO:2020-11-24 18:56:05,459:
Nov 24 18:56:05 "" python3[10163]: INFO:2020-11-24 18:56:05,806:
Nov 24 18:56:05 "" python3[10163]: HERE_WALLET_NAME =>
Nov 24 18:56:05 "" python3[10163]:         Addreses 
Nov 24 18:56:05 "" python3[10163]: INFO:2020-11-24 18:56:05,892: Obtaining bitcoin addresses to monitor . . .
Nov 24 19:01:44 "" python3[10163]: INFO:2020-11-24 19:01:44,100: Obtained list of addresses to monitor in 339.42725372314453sec
Nov 24 19:01:44 "" python3[10163]: INFO:2020-11-24 19:01:44,102: Building history with 4000 addresses . . .
Nov 24 19:01:46 "" python3[10163]: INFO:2020-11-24 19:01:46,764: Found 5 txes. History built in 0.6785974502563477sec
Nov 24 19:01:46 "" python3[10163]: INFO:2020-11-24 19:01:46,766: Starting electrum server
Nov 24 19:01:47 "" python3[10163]: Traceback (most recent call last):
Nov 24 19:01:47 "" python3[10163]:   File "/home/bitcoin/.local/bin/electrum-personal-server", line 11, in <module>
Nov 24 19:01:47 "" python3[10163]:     sys.exit(main())
Nov 24 19:01:47 "" python3[10163]:   File "/home/bitcoin/.local/lib/python3.5/site-packages/electrumpersonalserver/server/common.py", line 856, in main
Nov 24 19:01:47 "" python3[10163]:     run_electrum_server(rpc, txmonitor, config)
Nov 24 19:01:47 "" python3[10163]:   File "/home/bitcoin/.local/lib/python3.5/site-packages/electrumpersonalserver/server/common.py", line 538, in run_electrum_server
Nov 24 19:01:47 "" python3[10163]:     server_sock = create_server_socket(hostport)
Nov 24 19:01:47 "" python3[10163]:   File "/home/bitcoin/.local/lib/python3.5/site-packages/electrumpersonalserver/server/common.py", line 506, in create_server_socket
Nov 24 19:01:47 "" python3[10163]:     server_sock.bind(hostport)
Nov 24 19:01:47 "" python3[10163]: OSError: [Errno 98] Address already in use
Nov 24 19:02:48 "" python3[10275]: INFO:2020-11-24 19:02:48,403: Starting Electrum Personal Server
Nov 24 19:02:48 "" python3[10275]: INFO:2020-11-24 19:02:48,405: Logging to /tmp/electrumpersonalserver.log
Nov 24 19:02:48 "" python3[10275]: INFO:2020-11-24 19:02:48,968
```

OpenSystemCode, 2020-11-25:
I tried the suggestion by chris blecher change the port from 50002 to 50003 and the error didn*t appear. But why? Nothing listen to port 50002.
https://github.com/chris-belcher/electrum-personal-server/issues/45#issuecomment-414044344

Stadicus, 2020-12-13:
Thanks for this report, I'll try to reproduce. I assume using port `50003` works as a workaround for now?

Stadicus, 2021-11-27:
Sorry for never coming back on this issue. Unfortunately, EPS is no longer supported with the latest guide update to version 3. This is because Bitcoin Core needs to run with the wallet disabled to allow for proper access privileges. 

I recommend setting up Electrs, which is much more convenient once it's synced.

Closing this issue due for now. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus